sandeepmistry/arduino-BLEPeripheral #14: Blend Micro: Device not discoverable by Android BLE Controller app

AndrazP, 2015-01-11:
I tried BLEControllerSketch and SimpleChat example on Blend Micro, but I can't connect to the board with BLE Controller app for Android. The board is not discoverable.
I don't have this problems with Read bear lab examples.
I'm doing something wrong? I see pins are already set correctly for blend micro.

AndrazP, 2015-01-11:
The same story with LigthBlue app for Mac. With this library examples the device is not discoverable, RBL examples work fine.

sandeepmistry, 2015-01-11:
@AndrazP works for me when I open the serial monitor. Can you please try this?

I think the ```while(!Serial) {}``` needs to be removed.

AndrazP, 2015-01-11:
No, this is not a problem. Because I get "BLE UART Peripheral" string which is printed on the end of setup(); 
loop() is also working, but I can connect to the board because it's not visible.

sandeepmistry, 2015-01-11:
Weird, both examples are working for me, and I can see the device from my Mac (OS X 10.10) and iPhone (iOS 8.1.2).

Can you run we debug on, uncomment: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/nRF8001.cpp#L3

AndrazP, 2015-01-22:
Tried once more and it can be detected with every BLE app, with my Android app, only with Red bear lab BLE Controller app not. Thanks for great library!

closed 2015-01-22 by AndrazP

sandeepmistry, 2015-01-24:
@AndrazP thanks for the followup.

Just to clarify, you we're trying to use this example app? https://github.com/RedBearLab/Android/tree/master/Examples/BLEController

AndrazP, 2015-01-25:
Demo app from play store: https://play.google.com/store/apps/details?id=com.redbear.redbearbleclient

sandeepmistry, 2015-02-16:
@AndrazP I'm seeing the same thing with my Nexus 7 running 5.0 and the app. Couldn't see anything obvious in the [Android SDK](https://github.com/RedBearLab/Android) they open sourced.

However, their [iOS app](https://itunes.apple.com/us/app/ble-controller/id855062200?mt=8) detected the Blend Micro running the RBL contributed examples.