sandeepmistry/arduino-BLEPeripheral #16: Blend Micro: only a single characteristic being added

dmelancon, 2015-02-21:
I am trying to add two services to the Blend Micro.  It seems like the 2nd service always overwrites the first service.  When I use  advertisement-discovery.js on my central device, it will always only show the 2nd service added, or in the case of 3 services, it will show the last.  It works as should while using the Adafruit nRF8001 Breakout.

Code is here: https://gist.github.com/dmelancon/20cec5725943082d50cc

sandeepmistry, 2015-02-22:
@dmelancon it looks fine to me when I run it on my Blend Micro.

Are you concerned about only one service UUID being advertised? ``` blePeripheral.setAdvertisedServiceUuid``` should only be called once, if called multiple times the last one wins.

[image]

dmelancon, 2015-02-22:
This makes sense. I am seeing both now.  Thanks

closed 2015-02-22 by dmelancon

sandeepmistry, 2015-02-22:
@dmelancon no problem. Do you think the API docs and/or examples need to be updated to clarify anything?

dmelancon, 2015-02-22:
I think it is clear.  The actual cause of my problem ended up being the way I was looking for services on my central module running Noble.  I was running two separate peripheral.discoverServices as opposed to passing them as an array: peripheral.discoverServices(['fff0','fff3'], function(error, services) {});
The 2nd discovery would overwrite the first.