sandeepmistry/arduino-BLEPeripheral #17: Compile Error

KorkyraBoyCRO, 2015-03-12:
Hi...i have problem with verifying example codes like ancs etc
I placed the BLEPeripheral Library normally in my documents arduino library file and opened it in arduino IDE....after hitting verify button i got this error:

     /Users/myname/Documents/Arduino/libraries/BLEPeripheral/BLEProgmemConstantCharacteristic.cpp: In constructor 'BLEProgmemConstantCharacteristic::BLEProgmemConstantCharacteristic(const char*, const char*)':
     /Users/myname/Documents/Arduino/libraries/BLEPeripheral/BLEProgmemConstantCharacteristic.cpp:10: error: 'strlen_PF' was not declared in this scope

sandeepmistry, 2015-03-12:
@KorkyraBoyCRO more details please:

 * Which board are you using?
 * Which Arduino IDE version are you building with?
 * Which version of the library are you using (Git commit)?

KorkyraBoyCRO, 2015-03-13:
I am on the Uno R3 but i havent uploaded the code onto chip, i just tried to verify it ( i always first check it that way and if everything is ok i upload the code). Should i try to upload it ? As far as i know i have the latest IDE cause i had to reinstal arduino last month ( btw i am on the mac ) and for the library i also think it is last versio. I found it yesterday so i suppose it is the last

KorkyraBoyCRO, 2015-03-13:
I hope that ee can fix this cause i an really hoping ti get this run on arduino it look super cool and easy to work with code all in all it is awesome library

sandeepmistry, 2015-03-14:
@KorkyraBoyCRO a36a03a should fix this.

It was working with the new Arduino 1.6.x IDE, but not with the older 1.0.x IDE. In the future please provide exact version numbers, instead of using terms like the "latest".

closed 2015-03-14 by sandeepmistry