sandeepmistry/arduino-BLEPeripheral #18: Sharing SPI lines with second radio

andySigler, 2015-03-15:
Hello, and thank you for such an awesome library to get started with BLE!

I am connecting the nRF8001 (Bluefruit Le breakout) to Arduino Uno. Changed REQ pin to 8 and changed RDY pin to 3. Works great. I then add a second radio (RFm69HW), sharing SCK, MOSI, and MISO with the nRF8001. Without adding any new code to get the second radio working, I test the nordic radio, and it still works. I then import and use [this library](https://github.com/LowPowerLab/RFM69) to control the 2nd radio. Now, neither radio will initialize.

I have fallen back to the nRF8001 library [used here](https://github.com/tigoe/BLEDocs/tree/master/examples/nrf8001/led_service), and both radios can work together. It is only with BLEPeripheral that the SPI communication breaks down.

sandeepmistry, 2015-03-16:
@andySigler I'm surprised that combo works.

[arduino-nrf8001](https://github.com/guanix/arduino-nrf8001) uses the follow SPI settings:

```c
    // SPI mode 0; /16 clock divider
    SPI.setDataMode(SPI_MODE0);
    SPI.setBitOrder(LSBFIRST);
    SPI.setClockDivider(SPI_CLOCK_DIV16);
    SPI.begin();
```

[https://github.com/LowPowerLab/RFM69(https://github.com/LowPowerLab/RFM69/blob/master/RFM69.cpp) uses:
```c
  // save current SPI settings
  _SPCR = SPCR;
  _SPSR = SPSR;
  // set RFM69 SPI settings
  SPI.setDataMode(SPI_MODE0);
  SPI.setBitOrder(MSBFIRST);
  SPI.setClockDivider(SPI_CLOCK_DIV4); // decided to slow down from DIV2 after SPI stalling in some instances, especially visible on mega1284p when RFM69 and FLASH chip both present
  digitalWrite(_slaveSelectPin, LOW);
```

but it also saves and restores SPI registers (SPCR, SPSR). Which order are you calling begin etc?

You could also try manually storing/restoring the SPI register state, before/after blePeripheral calls.

sandeepmistry, 2015-04-04:
@andySigler I've pushed some changes to make the library use SPI transactions: 8b6dea47d86c60216bea0c37b981fdcff9b6ac06

Try them out, if you still have issues, please re-open.

closed 2015-04-04 by sandeepmistry