sandeepmistry/arduino-BLEPeripheral #19: Display caller ID, message title, message etc..

KorkyraBoyCRO, 2015-03-15:
First of all thank you for your help before. I hope i am not bothering you with  this.I want to ask u only is there any way to display caller ID, message text etc...? Oh yes...one more thing ...do u have an error when u try to compile ancs example with U8GLIB Library?

sandeepmistry, 2015-03-16:
@KorkyraBoyCRO could you please stick to one question/issue per Github issue, it makes tracking for me simpler. I've also setup [gitter](https://gitter.im/sandeepmistry/arduino-BLEPeripheral) for chat/questions.

What are you trying to get "display caller ID, message text etc..." from? Maybe share your sketch?

Please open a separate issue or use gitter for the U8GLIB question. With more details include:
 * Does the ANCS example compile by itself?
 * What is U8GLIB? Maybe a link?
 * What the compile error is.

KorkyraBoyCRO, 2015-03-17:
Hi...sorry for, lets say amateurism in this...So about ANCS. Yes it does compile by itself and it worked fine. I have connected arduino successfully to my iPhone 4s and received notification on Serial monitor. Only change i have done on you're original ancs example code is instead showing ancs category with number in serial monitor it now say incoming call, you got email and similar ( i did that with simple switch case statements)  . I was wondering is it possible to display full message with title if there is or caller ID when i got incoming call with your ancs example. I think i must use notification attributes  but not so sure and i don't see them...

P.S. i will try to upload a sketch as soon as possible...don't know how yet

KorkyraBoyCRO, 2015-03-17:
ok here it is...this is my sketch...as i told u  nothing much is really changed

https://gist.github.com/KorkyraBoyCRO/c3dd2ec1d8d79189ff31

sandeepmistry, 2015-03-19:
@KorkyraBoyCRO the ANCS example is pretty basic for now, it just receives the initial basic notification data, and does not query the iOS device for more information.

You would have to write some more code to do this, if you do get this working please submit a pull request for others to use. See [node-ancs](https://github.com/sandeepmistry/node-ancs) for a reference implementation on how to use ```ancsControlPointCharacteristic``` and ```ancsDataSourceCharacteristic``` to request more data from the iOS device for each notification.

KorkyraBoyCRO, 2015-03-23:
Ok i will try to get this working...thank you for your reply
One more thing, i know it is not related to this question, but i had noticed that ANCS example code takes a lot of global variables memory so i was thinking is it possible to get this library work with teensy 3.x since i got one but i am not sure how to do this?

sandeepmistry, 2015-03-25:
@KorkyraBoyCRO let's discuss the Teensy on Gitter ...

Btw, I'm closing this issue since the initial question is answered.

closed 2015-03-25 by sandeepmistry