sandeepmistry/arduino-BLEPeripheral #22: BLEPeripheral.begin function not executing

georgeedwards, 2015-06-22:
Having powered up my new nRF8001 module, it isn't showing up on any bluetooth devices ( I am primarily using the Bluefruit LE app recommended by ADA Fruit, on iPhone 5s, iOS 8.3 ).

Having traced the code, blePeripheral.begin (); isn't working in main.cpp (approx. line 68)

This is because nRF8001::begin isn't working in nRF8001.cpp

Because waitForSetupMode(); isn't working in nRF8001.cpp (approx. line 201)

I have traced this to the "while (!setupMode){" loop in nRF8001.cpp (approx. line 778)

Where and why does "setupMode" get activated is this a software issue, or is it triggered by a hardware input? I will continue tracing, but any thoughts would be much appreciated.

    bool lib_aci_event_get(aci_state_t *aci_stat, hal_aci_evt_t *p_aci_evt_data)
    {
      bool status = false;
  
      status = hal_aci_tl_event_get((hal_aci_data_t *)p_aci_evt_data);
  
      /**
      Update the state of the ACI with the 
      ACI Events -> Pipe Status, Disconnected, Connected, Bond Status, Pipe Error
      */
      if (true == status)
      {
        aci_evt_t * aci_evt;
      ...

What causes the “lib_aci_event_get(&this->_aciState, &this->_aciData” condition to be met?

don, 2015-06-22:
I'd try starting at a higher level.

What are you passing to BLEPeripheral()? Consult the [pinouts chart](https://github.com/sandeepmistry/arduino-BLEPeripheral#pinouts) for recommendations. For the Adafruit nRF8001, stick with the default BLE_REQ 10, BLE_RDY 2, BLE_RST 9 to start.

 What does your wiring look like? Does it match the pinouts? 

Can you get the [test example](https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/test/test.ino) to run on your hardware?

sandeepmistry, 2015-06-24:
Thanks @don!

@georgeedwards for further assistance, please provide answers to the questions Don asked in the previous comment.

georgeedwards, 2015-06-25:
Hi everyone, I have done some further fault finding and although the wiring was correct, my SPI ports on Mbed were faulty. As such, I have bought a new one and now have the library working! 

Thanks for your help.

sandeepmistry, 2015-06-25:
@georgeedwards great!

closed 2015-06-25 by sandeepmistry