sandeepmistry/arduino-BLEPeripheral #23: Arduino Freeze

mitko9000, 2015-07-22:
After I connect arduino zero + nrf8001 with any application it seems that the test sketch freeze after receiving the Evt Pipe Status.

Any sugestion how to handle this ?

zrecommerce, 2015-07-22:
Do you have any serial monitor output?

Alex Albino
Sr. Fullstack Developer
Zend Certified Engineer
https://github.com/zrecommerce
https://github.com/femtoduino
https://github.com/femtoio

(707) 246-6864

mitko9000, 2015-07-22:
Well what I did is to change nrf8001.cpp file inside case ACI_EVT_PIPE_STATUS
and commented everything now read and write is working.

mitko9000, 2015-07-22:
Also bonding for HID mouse is not working at all cant bond with device

sandeepmistry, 2015-07-25:
@mitko9000 I've addressed this with 051765c ... it didn't like dereferencing an unaligned uint64_t pointer, so I switched to using ```memcpy```.

closed 2015-07-25 by sandeepmistry