sandeepmistry/arduino-BLEPeripheral #25: Error compiling with red bear nano.

brigidrose, 2015-08-17:
The error message is as follows: 
/Documents/Arduino/libraries/BLEPeripheral/nRF51822.h:9:30: fatal error: s110/ble_gatts.h: No such file or directory
   #include <s110/ble_gatts.h>
                              ^
compilation terminated.
Error compiling.

sandeepmistry, 2015-08-18:
This occurs with the S130 branch of their Arduino board add-on: https://github.com/RedBearLab/nRF51822-Arduino

I have a patch that fixes the compile issues with some other necessary changes. However, need to discuss a few things with the RBL team first before sharing.

In the meantime, the suggested workaround is to uninstall the add-on from the board manager. Then follow the old instructions from their master branch: https://github.com/RedBearLab/nRF51822-Arduino/tree/master#install-nrf51822-arduino-add-on

sandeepmistry, 2015-08-22:
@brigidrose here the changes I've been working on for compatibility: https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/8c28140dee6a49ab0e48273d218f07cc456b544b

However, they depend on the following changes on the RedBearLab nRF51822 Arduino board add-on:
 * https://github.com/sandeepmistry/nRF51822-Arduino/commit/ab5f33b81777c78d412142567b1dc65d97893feb
 * https://github.com/sandeepmistry/nRF51822-Arduino/commit/b09717422c1b5d459e1487bfb7e441db50fcff5b

I've brought them up with the RBL team ...

sandeepmistry, 2015-12-06:
Pull request submitted: https://github.com/RedBearLab/nRF51822-Arduino/pull/41

Once this is merged by the RBL team, I will merge the following branch into master: https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/NRF51-S130

george-hawkins, 2015-12-20:
Yay - RBL have merged your pull request (commit [`e54b8fa`](https://github.com/RedBearLab/nRF51822-Arduino/commit/e54b8fa)) and released version 1.0.3 of there boards package for the Arduino IDE. So the current version now pulled in by the IDE includes your changes :smile:

I'm glad to see most of that stuff gone from `Arduino.h`. You could protect yourself against future potential name clashes by using namespaces. Unfortunately it's only with version 1.6.6 that `using namespace` works properly with `.ino` files - for earlier versions you're forced to qualify method parameter types explicitly with the appropriate namespace (due to the way the IDE "helpfully" automatically generates prototypes for all your functions so you don't need to forward declare them) :disappointed: 

Anyway in short please merge [NRF51-S130](https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/NRF51-S130) into master :smile:

sandeepmistry, 2016-01-17:
@george-hawkins sorry for the delay,  NRF51-S130 branch is merged into master now.

PR #48 (2016-01-17): Support for S130 via NRF51_S130 define
Resolves #25.
closed 2016-01-17 by sandeepmistry