sandeepmistry/arduino-BLEPeripheral #26: BondStore on RBL_nrf51822 does not write in a persistent way

bojh, 2015-10-13:
working with the new master branch (and the previous one), I have mentioned that the bond store is not written to the nrf81522 program memory. For the following log  I have inserted a hasData() in front of clearData() onto the BLEHIDPeripheral.clearBondStore() method:
Coupling with a central device (iOS App: BLE Tool) and switched on define NRF_51822_DEBUG, we can see following log:

    Evt Prim Srvc Disc Rsp 0x0
    Evt Prim Srvc Disc Rsp 0x10A
    Evt Sec Params Request 0 1 1 4 0 0 16
    Evt Conn Sec Update 1 2 16
    Evt Auth Status
    Evt Write, handle = 21
    01 00 
    Evt Write, handle = 25
    01 00 
    Evt Write, handle = 29
    01 00 
    Evt Write, handle = 33
    01 00 
    Evt Conn Param Update 0xC 0xC 0x4 0x12C
    Evt Disconnected
    Storing bond data
 
Additional question: Why is the bondstore written at the disconnect event and not before sending the BLEDeviceBonded() event in the BLE_GAP_EVT_AUTH_STATUS case?

Disconnect can be done by releasing the device in the iOS system BT configuration. At this moment I expect the bond data should be written. Rebooting my device calls clearBondstore() with a tweeked output with the bondstore.hasData() method:

    ClearBondStore!
    Bondstore: 0
    version = 6 89 73
    4
    1
    155
    Start advertisement
    BLE HID

... but the Bondstore data seems be not existent or written.

sandeepmistry, 2015-11-01:
@bojh I've corrected the issue, the bond store was dropping the first 4 bytes on retrieve, causing invalid data to be restored (https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/c2f4c1d9ad8df4213a7aa375ea21218d3d8aaf70).

> Additional question: Why is the bondstore written at the disconnect event and not before sending the BLEDeviceBonded() event in the BLE_GAP_EVT_AUTH_STATUS case?

This was done to match the nRF8001 implementation, however something we can discuss changing (in a new issue) ...

closed 2015-11-01 by sandeepmistry

bojh, 2015-11-02:
@sandeepmistry I've tested it well - thank's for your effort to track down that issue.
It enables me to use the security/bondstore functionality in the application.

bojh, 2015-11-02:
@sandeepmistry: It's OK for me to open a new issue for the bondstore write event.