sandeepmistry/arduino-BLEPeripheral #27: Can not discover device with HID examples

ayildirim, 2015-10-22:
Hi,

I have been trying all the samples. Scanning shows BLE device with ancs,start and test examples. But none of the other examples let me see BLE device on scans.

But when I upload HID or any other examples. Bluetooth device becomes undiscoverable. I can not see it no matter how long I wait.

I am using Arduino UNO with Adafruit BLE nrf8001.

Another thing; I could not discover BLE device on my MacBook with any of the examples.

ayildirim, 2015-10-22:
I tried putting Serial.println("passed Line 1...2..x") to debug.

It gets stuck at bleHID.begin();

ayildirim, 2015-10-22:
It gets stuck exactly at line below in the file BLEPeripheral.cpp

  this->_device->begin(advertisementDataType, advertisementDataLength, advertisementData,
                        scanDataType, scanDataLength, scanData,
                        this->_localAttributes, this->_numLocalAttributes,
                        this->_remoteAttributes, this->_numRemoteAttributes);

ayildirim, 2015-10-23:
it was lose soldering.

I strengthened soldering on the pins and it worked.

closed 2015-10-23 by ayildirim

sandeepmistry, 2015-10-27:
@ayildirim great, thanks for the updates and for closing.