sandeepmistry/arduino-BLEPeripheral #28: Sketch size is too big for Arduino UNO

ayildirim, 2015-10-24:
Hi!

I'm trying to optimise library to reduce the size of sketch and libraries. But so far, it is not going well.
Any suggestions on this will be much helpful.

I disabled DEBUG printouts.

What else can I do?

Thanks,

ayildirim, 2015-10-24:
I will use two Arduino's as a solution. 1 arduino to listen from serial port for commands to be issued on BLE module, 2nd arduino to handle rest of the stuff.

Luckily I had some Arduino Mini pro laying around :dancers:

ayildirim, 2015-10-24:
I think I'm experiencing run-time memory problems. 

I change completely unrelated part of the code. Then suddenly, it starts getting stuck at begin call for bluetooth module.

For example, I define a variable object(BLEMultimedia). Then suddenly it starts getting stuck. Even if I don't use that variable object.

ayildirim, 2015-10-25:
When I have this memory problem it gets stuck at below section in DEBUG print out.

Evt Device Started: Setup
07 06 00 00 03 02 41 FE 
Evt Cmd Rsp: Transaction Continue
1F 06 10 00 00 00 00 00 02 00 07 00 08 01 01 00 00 06 00 06 00 00 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
1F 06 10 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 50 00 03 90 00 FF 
Evt Cmd Rsp: Transaction Continue
1F 06 10 38 FF FF 02 58 0A 05 00 00 40 00 00 00 40 00 00 00 50 00 00 00 40 00 01 02 01 02 01 02 
Evt Cmd Rsp: Transaction Continue
05 06 10 54 01 02 
Evt Cmd Rsp: Transaction Continue
19 06 70 00 02 02 12 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
19 06 70 16 09 0A 48 49 44 20 56 6F 6C 75 6D 65 00 00 00 00 00 00 00 00 00 00 
Evt Cmd Rsp: Transaction Continue
0E 06 20 00 04 04 02 02 00 01 28 00 01 00 18 
Evt Cmd Rsp: Transaction Continue
0F 06 20 0B 04 04 03 03 00 00 28 03 01 02 01 00

ayildirim, 2015-10-25:
My tests show, it is able to finish beginning process if memory left for local variable is at least 515 bytes.

It stops working, if you have less than 515 bytes left for local variables.

don, 2015-10-25:
@ayildirim memory management is tricky, I often run out of memory when creating peripherals. The Leonardo based boards have 512bytes more SRAM than an Uno. The nRF51822 boards (RedBear and RFduino) have 16 KB of SRAM.

I also use the `F()` macro to get as many of my strings as possible stored in flash instead of RAM.

ayildirim, 2015-10-25:
@don yes. I think I will have to go with Teensy or something. I was going to make a custom gamepad/tv controller.

I thought using 2 arduinos. But there isn't even enough space left for communication on arduino handling nrf8001.

sandeepmistry, 2015-10-27:
@ayildirim are you concerned about code space or RAM?

The library is pretty ambitious on the nRF8001 side, it builds the nRF8001 setup messages on the fly on the Arduino (Nordic's Arduino lib relies on a desktop app for this) . There might be some optimizations, but would make the code harder to maintain so I stopped pursuing them.

Happy to review any suggestions you have though.

ayildirim, 2015-10-27:
Hi @sandeepmistry ,

How much memory I can save by sending setup messages on the fly?

Is it only that setup messages array in nrf8001.cpp? 
Or is there more added to it in runtime somewhere?

I will try to implement same thing your code, if it can save enough memory. Otherwise Im gonna go ahead and buy a Teensy 3.1

sandeepmistry, 2015-10-27:
@ayildirim 

> How much memory I can save by sending setup messages on the fly?

You mean by not calculating them on the fly? No idea ...

> Is it only that setup messages array in nrf8001.cpp? 
Or is there more added to it in runtime somewhere?

Many more added at run time ...

> I will try to implement same thing your code, if it can save enough memory. Otherwise Im gonna go ahead and buy a Teensy 3.1

Good luck!

I would say go for the Teensy or use this lib with nRFgo studio: https://github.com/NordicSemiconductor/ble-sdk-arduino/tree/master/documentation/libraries/BLE/examples/ble_HID_template

probably will save you a ton of time instead of modding this lib.

daviddedwin, 2015-10-28:
If you are running out of Flash space during the prototyping stage, it may be worthwhile to switch the ble SDK for Arduino from Nordicsemi which will generate the Setup in the nRFgo Studio (Windows tool and works on the OS X using playonmac), this will allow you to skip the library used to build the Setup in the arduino-BLEPeripheral. The Setup messages are generated by the nRFgo studio and are stored in Flash, so this needs about 1 to 2Kbytes of Flash.

You may be able to cut out the Setup part of the code from the arduino-BLEPeripheral and use the rest but I will have to leave that to Sandeep to comment on.

ayildirim, 2015-10-28:
I need to use Mouse, Keyboard HID features together on 1 device. This library allows doing that. I am not sure if it is possible with nrfGO Studio. It compiles setup message with a given xml descriptor file for either keyboard or mouse.

don, 2015-10-28:
This library is way nicer to use than nRFgo. Spending $19 on a Teensy is the easiest/fastest solution.

daviddedwin, 2015-10-28:
@ayildirim , combining mouse and keyboard functionality is pretty straightforward on nRFgo Studio as well. Additionally you can do the combining on this library and then read the GATT Server over the air and replicate it on the nRFGo studio. 
Keeping your RAM and Flash needs low helps you to have choices on mcus. however if that is not needed or there is no real pricing pressure for you, you can pick a solution that has the RAM and flash needed available.

sandeepmistry, 2015-10-29:
@daviddedwin @don thanks for the feedback/comments!

@ayildirim can we close this now?

ayildirim, 2015-10-29:
@sandeepmistry yes. Thank you all for your answers. Thank you for great work @sandeepmistry.
I will take @don 's advice and continue with Teensy.

closed 2015-10-29 by ayildirim