sandeepmistry/arduino-BLEPeripheral #34: Add support for the Bluefruit LE Micro

jonblack, 2015-11-14:
I've been trying to get this library to work with the BlueFruit LE Micro board (https://www.adafruit.com/products/2661) but haven't had much luck. It's not in the supported list, so this isn't a huge surprise, although it using the nRF51822 ble chip.

don, 2015-11-14:
Unfortunately the new nRF51 based Adafruit boards don't work with this library.

sandeepmistry, 2015-11-15:
@jonblack the BlueFruit LE Micro boards has a bit of a different architecture than the nRF51822 boards this library supports. This library runs directly on the ARM M0 processor on the nRF51822.

Where as the Bluefruit board has it's own f/w running on the nRF51 with the ATmega32u4 running as master. Adafruit does have a series of [AT commands](https://learn.adafruit.com/bluefruit-le-micro-atmega32u4-microcontroller-usb-bluetooth-le-in-one/at-commands) to interact with their nRF51 firmware, however it doesn't seem to cover the entire BLEPeripheral API. So supporting the board is not high on the priority list.

Having said that, if you have time to create a new ``BLEDevice`` implementation for the AT commands, I'd be happy to merge a pull request for it.

vojtamolda, 2015-11-18:
Hello @jonblack,

I'll be more than happy to help you to write/test the new `BLEDevice` implementation using Adafruit AT commands, if you plan to do it. I would love to see BLEPeripheral supporting *Bluefruit LE Micro* (and possibly *Bluefruit LE SPI/UART Friend* + *Arduino*) boards.

sandeepmistry, 2015-11-19:
@vojtamolda awesome! Please keep us posted on you progress here :)

vojtamolda, 2015-11-20:
So far my development [branch](https://github.com/vojtamolda/arduino-BLEPeripheral/tree/BluefruitAT) is progressing well. I haven't done any BTLE development before, so first I'm trying to familiarize myself with the codebase and BTLE in general.

The biggest problem I see is that the Adafruit AT command set doesn't support BTLE central role. Unless they decide to add support for it (or open-source the firmware running on the nRF51), there's no way how to implement any of the `*RemoteCharacteristic(...)` methods in `BLEDevice`.

I'll try to write a comment here once (and if :-) my efforts will lead to something meaningful.

sandeepmistry, 2015-11-20:
@jonblack cool, this library doesn't support central role either. However, peripherals can discover services and characteristics on the connected central (it's how the iOS ANCS protocol works, service solicitation). Which is what the library supports, and they are not super key for most peripherals.

> I'll try to write a comment here once (and if :-) my efforts will lead to something meaningful.

Sounds good! Setting up the advertisement data, and the services and characteristics are the key pieces.

jonblack, 2015-11-20:
@vojtamolda Great to hear you're giving it a shot. I'd love to help but I'm swamped at the moment. If you need help testing, I reckon I can fit that in.

vojtamolda, 2015-12-07:
I think I've changed my mind about this whole thing. I've gotten reasonably far with my [efforts](https://github.com/vojtamolda/arduino-BLEPeripheral/tree/BluefruitAT) to bridge *Adafruit*'s AT commands into *BLEPeripheral* API. Two most simple example sketches (test.ino and starter.ino) are working, but with some limitations.

At this point I don't think it makes sense to carry on with the project. There are two main reasons I see:

1. The number of features not supported by *Adafruit* AT commands is too high for a meaningful functionality. Here's a list:
  - There's a lot of services UUID=0x1800`  (Generic Access Profile) and characteristics `UUID=0x2A00` (Device Name), `0x2A01` (Appearance), `0x2A05` (Service Changed), that are always on and can't be changed. Firmware (v0.6.7) responds with *OK*, but doesn't perform the requested change and continues to broadcast old values.
  - No AT command to access remote characteristics on the central device 
  - No AT command to access to handle bonding storage
  - No AT support for descriptors

2. [Bluefruit Micro LE](http://www.adafruit.com/products/2661) is marked as discontinued and replaced by [Bluefruit Feather LE](https://www.adafruit.com/products/2829). *Feather* unfortunately uses very similar architecture (*ATmega 32u4* master and *nRF51* slave) and same set of AT commands. Since *Micro* was released in August, this is really a surprise for me.

*Adafruit* also doesn't seem to plan to make their *nRF51* [firmware](https://github.com/adafruit/Adafruit_BluefruitLE_Firmware) open source so there's very little chance to add support/change any of the above.

sandeepmistry, 2015-12-12:
@vojtamolda great effort!

> There's a lot of services UUID=0x1800 (Generic Access Profile) and characteristics UUID=0x2A00 (Device Name), 0x2A01 (Appearance), 0x2A05 (Service Changed), that are always on and can't be changed. [...]

Fair enough, can the Device name be changed?

> No AT command to access remote characteristics on the central device
> No AT command to access to handle bonding storage

Good to know, I wouldn't consider these essentials myself.

> No AT support for descriptors

This one I would consider essential.

@jonblack @don what do you think? Should we close this?

jonblack, 2015-12-13:
@vojtamolda Great research. It's a real shame that Adafruit hide the potential of the Bluetooth LE chip behind such a limited API.

I agree that it's best to stop trying to support it.

vojtamolda, 2015-12-13:
Thanks @jonblack and @sandeepmistry. It hardly makes any difference now, but device name can be changed. (Unless you've created your own `UUID=0x2A00` characteristic, which prevents the corresponding (and buggy) AT command from working... :-)

sandeepmistry, 2015-12-13:
Alright, let's close this for now.

@vojtamolda, thanks for looking into this again!

closed 2015-12-13 by sandeepmistry

don, 2015-12-14:
I think the biggest problem is not getting a callback when a central updates a characteristic value. Polling characteristics for changes stinks. Kevin Townsend said it might be possible to get an IRQ with the [SPI version of the Bluefruit LE](https://www.adafruit.com/products/2633) but I haven't tried this yet.

Unfortunately, abandoning this probably makes sense. :(

larsgk, 2015-12-15:
I know this is closed - but could it make sense to e.g. solve this in collaboration with Adafruit? I am sure they/she would be interested in a good solution and I think the BLE part can be firmware updated (to fixthe issues)

vojtamolda, 2015-12-15:
Hello @larsgk,

I think you're right. Solving this in collaboration with Adafruit is the only reasonable option and it's definitely possible with OTA updates.

If they decide to add new AT commands, I'm willing to give it another shot, since I still have a couple of unused *Bluefruit Mircro*s laying around ;)

sandeepmistry, 2015-12-20:
@larsgk @vojtamolda feel free to get the conversation going with Adafruit, I can make introductions if necessary. I still think a binary based API would be better than AT commands.

Has anyone experimented with flashing alternative firmware on the board? The Nordic serialization API would be a much better fit than the closed source Adafruit AT command set. https://developer.nordicsemi.com/nRF51_SDK/nRF51_SDK_v8.x.x/doc/8.1.0/s130/html/a00111.html

gto406, 2016-02-18:
Hi Sandeep, I noticed I can flash the LE via the AdaFruit BLE (Able iOS) app.  They also have a mechanism (in Python) that uses a Segger or a Rasp-Pi to update.  I was not sure however, if the firmware formats (hex files) had a specific breakout (boot-loaded, etc.)

Do you recommend any Firmware I can start with for their Bluefruit LE Shield setup on an Arduino Mega (Nordic nRF51822)?  I am hoping to get a version of BLEPeripheral going as the AdaFruit firmware IMHO is severely limited.

Thanks!
Brian.