sandeepmistry/arduino-BLEPeripheral #38: "Unsupported platform" error with BLE Nano

ncr, 2015-11-29:
I am using the BLE Nano board with nRF51822 Arduino Add-on from https://github.com/RedBearLab/nRF51822-Arduino and can compile the "blink" example without problems for the BLE Nano board.

However, when trying to `#include <BLEPeripheral.h>` I'm getting the following:

```
In file included from /Users/ncr/Documents/Arduino/libraries/BLEPeripheral/utility/lib_aci.h:34:0,
                 from /Users/ncr/Documents/Arduino/libraries/BLEPeripheral/nRF8001.h:6,
                 from /Users/ncr/Documents/Arduino/libraries/BLEPeripheral/BLEPeripheral.h:23,
                 from /Users/ncr/dev/blego/blego/blego.ino:3:
/Users/ncr/Documents/Arduino/libraries/BLEPeripheral/utility/hal_platform.h:75:6: error: #error "Unsupported platform"
     #error "Unsupported platform"

      ^
exit status 1
Error compiling.
```

The complete code I'm trying to compile:

```
#include <SPI.h>
#include <BLEPeripheral.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Is BLE Nano supposed to work stand-alone or only as an add-on to an Arduino UNO (or similar) board?

sandeepmistry, 2015-12-01:
@ncr 

> Is BLE Nano supposed to work stand-alone or only as an add-on to an Arduino UNO (or similar) board?

It's suppose to work stand-alone.

I can reproduce the error with Arduino IDE 1.6.6, but 1.6.5 is fine:

1.6.5
```/Users/smistry/Library/Arduino15/packages/RFduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -DBLE_STACK_SUPPORT_REQD -DDEBUG_NRF_USER -DNRF51 -DBOARD_PCA10001 -mcpu=cortex-m0 -DF_CPU=16000000 -DARDUINO=10605 -mthumb -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/cores/RBL_nRF51822 -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/variants/BLE_Nano -I/Users/smistry/Documents/Arduino/libraries/MemoryFree -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/libraries/SPI -I/Users/smistry/Documents/Arduino/libraries/BLEPeripheral -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/system/CMSIS/CMSIS/Include -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/app_common -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/ble -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/ble/ble_services -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/ble/rpc -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/gcc -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/s110 -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/sd_common -I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/nrf51822_SDK/Include/sdk /var/folders/5d/pmvtw7p579n5npm6p421qf540000gn/T/build3721415366949437813.tmp/test.cpp -o /var/folders/5d/pmvtw7p579n5npm6p421qf540000gn/T/build3721415366949437813.tmp/test.cpp.o ```

1.6.6
```"/Users/smistry/Library/Arduino15/packages/RFduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -w -x c++ -M -MG -MP -DF_CPU=16000000 -DARDUINO=10606 -DARDUINO_RBL_NRF51822_NANO -DARDUINO_ARCH_RBL_NRF51822  -mthumb "-I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/cores/RBL_nRF51822" "-I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/variants/BLE_Nano" "-I/Users/smistry/Documents/Arduino/hardware/RBL/RBL_nRF51822/libraries/SPI" "-I/Users/smistry/Documents/Arduino/libraries/BLEPeripheral" "/var/folders/5d/pmvtw7p579n5npm6p421qf540000gn/T/build8ed0b53cabcea151c31d778349f0b8b7.tmp/sketch/test.ino.cpp"```

Something has changed between IDE versions, ``` -DNRF51``` is no longer passed in. BLEPeripheral use this define to determine nRF51822 or nRF8001 compile mode.

It is set in ```platforms.txt``` here: https://github.com/RedBearLab/nRF51822-Arduino/blob/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/platform.txt#L20

A workaround for now is to use 1.6.5 of the IDE until this is resolved in this library or the RBL core.

ncr, 2015-12-01:
Thanks for the info, this made me happy, BLEPeripheral API is much nicer than the rest :) I'll try to debug this issue, though my experience is a bit lacking.

ncr, 2015-12-01:
I have tried Arduino 1.6.7 hourly build and the issue disappears. However, to compile successfully I had to add `/core/` to the paths used by the combining step of compilation: https://github.com/ncr/nRF51822-Arduino/commit/2ad87fe2a748f939e953348c409210072920339f.

sandeepmistry, 2015-12-01:
@ncr awesome progress so far!

Any suggestions on how to move forward? Are you planning on submitting a pull request to RBL?

What error do you get without using your forked version?

ncr, 2015-12-02:
Without the patch, the following lengthy line fails because both `syscalls.c.o` and `startup_nrf51822.c.o` are placed in the `{build.path}/core` directory.
```
"/Users/ncr/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -Os -Wl,--gc-sections -mcpu=cortex-m0 -mthumb "-T/Users/ncr/Documents/Arduino/hardware/RBL/RBL_nRF51822/variants/BLE_Nano/linker_scripts/gcc/RBL_nRF51822.ld" --specs=nano.specs "-Wl,-Map,/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/blego.ino.map" -o "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/blego.ino.elf" "-L/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp" -lm -lgcc -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/syscalls.c.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/sketch/blego.ino.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/SPI/SPI.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEAttribute.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEBondStore.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLECentral.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLECharacteristic.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEConstantCharacteristic.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEDescriptor.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEDevice.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEFixedLengthCharacteristic.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEHID.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEHIDPeripheral.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEHIDReportMapCharacteristic.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEHIDReportReferenceDescriptor.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEKeyboard.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLELocalAttribute.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEMouse.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEMultimedia.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEPeripheral.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEProgmemConstantCharacteristic.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLERemoteAttribute.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLERemoteCharacteristic.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLERemoteService.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEService.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLESystemControl.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLETypedCharacteristics.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEUtil.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/BLEUuid.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/EddystoneBeacon.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/iBeacon.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/nRF51822.cpp.o" "/var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/libraries/BLEPeripheral/nRF8001.cpp.o" "/va [... cut ...] m_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/core/core.a" -Wl,--end-group
```

```
arm-none-eabi-g++: error: /var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/syscalls.c.o: No such file or directory
arm-none-eabi-g++: error: /var/folders/6w/rlm_5ddx4tx4qc_b1dlcx9zm0000gn/T/build5cf3396e0223707169cc3a6f5fd954d1.tmp/startup_nrf51822.c.o: No such file or directory
```

The PR: https://github.com/RedBearLab/nRF51822-Arduino/pull/39

sandeepmistry, 2015-12-04:
@ncr thanks!

I belive RBL is no longer supporting the master branch and S130 is published via the board manager. @Cheong2K is this correct?

Cheong2K, 2015-12-04:
We support S130 branch and will add to the board manager very soon for the 1.6.6 IDE.

ncr, 2015-12-04:
@sandeepmistry Glad to be of help!

closed 2015-12-04 by ncr

KTibow, 2020-05-08:
I'm still getting this. I'm trying to use GitHub actions with [this workflow](https://github.com/family-richards/Napkin/blob/master/.github/workflows/main.yml). Does anybody know how to fix this problem?

KTibow, 2020-05-08:
Never mind, I was running code in the examples, I just removed that code.