sandeepmistry/arduino-BLEPeripheral #39: Please provide EddystoneTLM

iamdenny, 2015-11-30:
I just checked your examples which is https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/master/examples/Eddystone
There are only UID and URL examples.
I would like to get TLM example.
Please make it for me.
Thanks.

sandeepmistry, 2015-12-01:
@iamdenny it would be great if you had time to submit a pull request for this!

A few pointers:
 * https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/EddystoneBeacon.cpp
 * https://github.com/google/eddystone/tree/master/eddystone-tlm

In ```EddystoneBeacon::loop()```, a check would need to be added to switch the Eddystone data advertised from UID/URL to TLM at a regular interval. For example, current advertisement interval is 100ms, so TLM data could be interleaved every 10 packets or so.

arun271, 2016-03-30:
@sandeepmistry  @iamdenny  Please check this and let me know if it helps.
https://github.com/arun271/arduino-BLEPeripheral/blob/eddystone/examples/Eddystone/EddystoneTLM

Thanks.