sandeepmistry/arduino-BLEPeripheral #40: Board specifics that need porting

jeremytregunna, 2015-12-06:
I'm currently looking through the code, with the aim of porting this library to run on the STM32F4-Discovery board with an nRF8001. From what I can see, the only changes I need to make are pins, but I'm not sure this is entirely accurate. Guidance would be great in the lieu of porting documentation.

sandeepmistry, 2015-12-06:
@jeremytregunna 

If the STM32F4 supports the Arduino SPI library API, it's pretty simple. Just follow what I did to add Teensy support: https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/decf3277bb3e4fc708dc03bb98f81722dce42208#diff-b974adcbcffc1dac0599562a46e40eff

The nRF8001 ACI code is originally from the Nordic Arduino BLE SDK, so the following guide maybe be of interest as well. Note, I have remove the interrupt driven code in this library:

https://github.com/NordicSemiconductor/ble-sdk-arduino/blob/master/documentation/libraries/BLE/nRF8001-Porting-ACI-Library.md#getting-the-nrf8001-sdk-for-arduino-ready-for-porting

jeremytregunna, 2015-12-06:
@sandeepmistry Ok thank you very much.

closed 2015-12-06 by jeremytregunna

sandeepmistry, 2015-12-06:
@jeremytregunna no problem.

Pull request welcome once you get it working :)

jeremytregunna, 2015-12-06:
Definitely :)