sandeepmistry/arduino-BLEPeripheral #41: Question: Does anyone know of a "blend micro on steriods".

linuxdevel, 2015-12-15:
The Blend Micro just has ~28KB flash available which is very little if you want to use it for something a bit more serious than just playing around.. :)
It would have been nice to have a "blend micro variant" with ATmega64/128 or even 256... :) (With similar small physical footprint)

Anyone who knows of such a device?

Arne

larsgk, 2015-12-15:
Perhaps http://redbearlab.com/blenano/

linuxdevel, 2015-12-15:
I know about BLE Nano. But with the USB interface it is "much" bigger. And USB is required for host communication.
The blend Micro formfactor is ideal for embedding into a small box.. 😊

larsgk, 2015-12-15:
I think you only need the USB board connected when programming ;)

linuxdevel, 2015-12-15:
Yes for sure.. However, the device will be partly controlled from a host computer through USB as well as automated sw upgrades. So I need the USB connectivity in a small form factor just like the micro :)

daviddedwin, 2015-12-16:
Perhaps a Teensy + ( [Tiny circuits](https://www.tiny-circuits.com/tiny-shield-bluetooth-low-energy-nordic.html) / [Jaycon systems](http://www.jayconsystems.com/nrf8001-development-board.html) / Adafruit LE (https://www.adafruit.com/products/1697) ).

sandeepmistry, 2015-12-20:
@linuxdevel do you want to stick with an AVR MCU are you open to ARM?

This library will most likely be supporting the new @redbearlab Duo in the future: https://www.kickstarter.com/projects/1991736672/redbear-duo-a-small-and-powerful-wi-fi-ble-iot-boa

cc/ @Cheong2K

linuxdevel, 2015-12-21:
@sandeepmistry the new redbear duo looks promising, but maybe the unit price will be "very high"? :)

Cheong2K, 2015-12-22:
@linuxdevel 

The Duo price will not be "very high", we will accept new orders from Jan. 2016, it is a board with WiFi + Bluetooth 4.1 (Smart Ready), please visit www.redbear.cc for details.

linuxdevel, 2015-12-22:
@Cheong2K 

The Duo sure looks promising, Perfect physical footprint and enough flash. Just waiting for it to be available :)

sandeepmistry, 2015-12-23:
@linuxdevel can we close this issue now?

sandeepmistry, 2016-01-03:
Closing due to lack of activity.

closed 2016-01-03 by sandeepmistry