sandeepmistry/arduino-BLEPeripheral #42: broadcasting an Eddystone URL and running a GATT service simultaneously via RFDuino

matthewsibigtroth, 2015-12-15:
Hello!
In the code below I am attempting to use EddystoneBeacon as both a URL broadcaster and a GATT service on an RFDuino. However it appears that the line

`eddystoneBeacon.setAdvertisedServiceUuid(ledService.uuid());`

prevents the URL from being broadcast. If I comment it out, the URL broadcasts but the service is not discoverable (the latter makes sense, it being an advertisement call). Curious if this has been experienced before, or if this isn't the proper way to combine the two features?
Thanks!

```
#include <SPI.h>
#include <EddystoneBeacon.h>

#define EDDYSTONE_BEACON_REQ   6
#define EDDYSTONE_BEACON_RDY   7
#define EDDYSTONE_BEACON_RST   4
EddystoneBeacon eddystoneBeacon = EddystoneBeacon(EDDYSTONE_BEACON_REQ, EDDYSTONE_BEACON_RDY, EDDYSTONE_BEACON_RST);
BLEService ledService = BLEService("19b10000e8f2537e4f6cd104768a1214");
BLECharCharacteristic switchCharacteristic = BLECharCharacteristic("19b10001e8f2537e4f6cd104768a1214", BLERead | BLEWrite);

void setup() {
  initBeacon();
}

void initBeacon() {
  Serial.begin(115200);
  #if defined (__AVR_ATmega32U4__)
  delay(5000);  //5 seconds delay for enabling to see the start up comments on the serial board
  #endif

  eddystoneBeacon.setConnectable(true);
  eddystoneBeacon.setLocalName("CANDY BOY");
  eddystoneBeacon.setAdvertisedServiceUuid(ledService.uuid());
  eddystoneBeacon.addAttribute(ledService);
  eddystoneBeacon.addAttribute(switchCharacteristic);
  eddystoneBeacon.begin(-18, "https://example.com");
}

void loop() {
  eddystoneBeacon.loop();
}
```

sandeepmistry, 2015-12-20:
@matthewsibigtroth this is a limitation of the ```BLEPeripheral``` API only one service UUID can be advertised. This was done to keep things simple on the nRF8001 initially, as well as the API.

The Eddystone service UUID is set in the constructor of ```EddystoneBeacon```, maybe it should be moved to ```begin``` so user's can't override it.

https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/EddystoneBeacon.cpp#L39

sandeepmistry, 2016-01-03:
@matthewsibigtroth anything to further discuss on this?

matthewsibigtroth, 2016-01-04:
Thanks Sandeep. This makes sense and was honestly an oversight on my part. I'll make the added GATT service UUID match up with that of the Eddystone beacon and I believe that should allow for both the GATT service AND the URL broadcasting to be simultaneously functional. In short, nothing further to discuss. :)

sandeepmistry, 2016-01-05:
@matthewsibigtroth no problem.

I've pushed a change (https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/e381d2581a2a17cae20b416ddb9824230fd714dc) to prevent overriding the advertised service UUID if EddystoneBeacon is used.

closed 2016-01-05 by sandeepmistry