sandeepmistry/arduino-BLEPeripheral #43: GCC warnings

linuxdevel, 2015-12-16:
When compiling I am used to use -Wall -Werror as two of the options to gcc to force me to write warning free code (and hopefully/eventually more error free code).  If there are warnings that I decide not to handle, then I either use pragma's to switch them off, or use macro's to trick the compiler to ignore them.

The arduino-BLEPeripheral lib gives quite alot of warnings. (unused parameters, no return statement in function returning non-void, enumeration value  not handled in switch etc)

Are there any plans to reduce number of warnings in the library?

Arne

sandeepmistry, 2015-12-20:
@linuxdevel thanks for bringing this up, the Arduino IDE does not have ```-Wall -Werror``` enabled by default, so I missed these.

I've taken a first cut at the warnings in this branch, I still need to test on both nRF8001 and nRF51822 boards: https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/gcc-warnings

sandeepmistry, 2016-01-03:
@linuxdevel have you had a chance to look at my branch?

linuxdevel, 2016-01-03:
Hello Sandeep, Happy New Year :)

No I havent checked out your branch, however, if you have tested it ok. It would be good to merge it  into master.

sandeepmistry, 2016-03-06:
Done in 026e60238bba45288a0dc230cebfcca085abd3d3.

closed 2016-03-06 by sandeepmistry