sandeepmistry/arduino-BLEPeripheral #44: TX Power Level Bluetooth Characteristic

linuxdevel, 2015-12-16:
Hi

Is there a way to make the nRF8001 device to broadcast  the TX Power Level Bluetooth Characteristic? 

Arne

linuxdevel, 2015-12-17:
If I'm not wrong, I can just add a characteristics according to https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=239393

Question is answered.. Feel free to delete the question.

daviddedwin, 2015-12-17:
@sandeepmistry Not sure how this is done in the API, the nRF8001 ACI has the Broadcast/Connect/Bond all of which can be used to advertise data. The ACI command OpenAdvPipe is also used to load data from a Specific Characteristic into the advertising data. The [Temperature broadcast](https://github.com/NordicSemiconductor/ble-sdk-arduino/tree/master/libraries/BLE/examples/ble_temperature_broadcast) example shows how to broadcast a Characteristic.

sandeepmistry, 2015-12-20:
@linuxdevel do you want to advertise the TX power using GAP? (This is not supported).

@daviddedwin characteristics can be broadcasted in the GAP service data using the ```characteristic.broadcast()``` API. For more info see:
 * https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#broadcast
 * EddystoneBeacon:
   * https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/EddystoneBeacon.cpp#L34
   * https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/EddystoneBeacon.cpp#L68

linuxdevel, 2015-12-21:
@sandeepmistry  I thought it was supported from what I see here : https://www.bluetooth.org/en-us/specification/assigned-numbers/generic-access-profile

0x0A	«Tx Power Level»	Bluetooth Core Specification:
Vol. 3, Part C, section 8.1.5 (v2.1 + EDR, 3.0 + HS and 4.0)
Vol. 3, Part C, sections 11.1.5 and 18.3 (v4.0)
Core Specification Supplement, Part A, section 1.5

sandeepmistry, 2015-12-23:
@linuxdevel yep, it's part of the spec.

However, BLEPeripheral tries to keep things simple.

It only allows the following:
 * Advertisement data: service UUID or Manufacturer data + service data for broadcasted characteristic (if there is space)
 * Scan response data: local name

sandeepmistry, 2016-01-03:
@linuxdevel I'm closing this for now.

closed 2016-01-03 by sandeepmistry