sandeepmistry/arduino-BLEPeripheral #47: call setLocalName() with BLEDevice address before advertising

dakoner, 2016-01-16:
I am porting an Arduino code that currently uses the Nordic nRF8001 SDK to use arduino-BLEPeripheral.  
Our firmware sets the visible name to include part of the local device's address. We achieved this using the Nordic SDK by requesting the address, waiting for the address to be available while pumping the event stack, and capturing the address in setup phase, then starting advertising with that name.

I tried to do this via BLEPeripheral and hit several snags: 
0) I tried to do it through vanilla API actions, but couldn't.  It seems like you can't access the device address until BLEPeripheral.begin() is called, and you can't change the local name/re-advertise it after begin() is called.

1) the BLEPeripheral doesn't expose the local device address to users.  I worked around this by defining a subclass of BLEPeripheral that handles BLEDeviceAddressReceived and saves the address as an instance variable.

2) Even then, I can only do peripheral.setLocalName(value) before peripheral.begin().  Calling it after has no effect, at least it does not seem to change the name that is visible when scanning the device.

So basically my question is:
1) is it possible to do this via the vanilla API (access the device local address, and use it to set the local name, before advertising)?
2) if not, what is the right way to achieve this?

sandeepmistry, 2016-01-17:
@dakoner 

> 1) is it possible to do this via the vanilla API (access the device local address, and use it to set the local name, before advertising)?

It's a bit ugly, but it might be possible with 2 calls to ```begin```.

1. Call ```begin``` once
2. Get local address
3. Set local name based on address
4. Call ```begin``` again.

> 2) if not, what is the right way to achieve this?

This is not a use case this library had in mind.

The library to could be modified to initialize SPI communications and reset the nRF8001 prior to ```begin``` if the local address request is requested.

What Nordic nRF8001 SDK API were you using to update the local name at runtime? I'm wondering if this technique could be used in this library ....

dakoner, 2016-01-17:
Thanks, sandeep.   I already tried your idea of calling begin() twice and the sketch hung on the second call to begin().  I'll try this again to verify that for certain it's hanging the sketch.

The code we used with the nRF8001 SDK API isn't open source, but I can describe what we did (and we can also open source the file if it helps).  I can't be 100% certain any of the following steps is truly required to successfully change the name.

1) in our own begin() routine, we setup aci_state and call lib_aci_init
2) in the Poll loop, for ACI_EVT_DEVICE_STARTED, mode=ACI_SETUP, we set setup_required=true
3) at the end of the poll loop, if setup_required==true, we call do_aci_setup(), and if there is no error, set setup_required=false
4) in poll loop, for ACI_EVT_DEVICE_STARTED, mode=ACI_STANDBY, we set addrRetrieved=false, lib_aci_get_address(), then call poll in a loop until addrRetrieved=true, 
5) in poll loop, for ACI_EVT_CMD_RSP, when ACI_CMD_GET_DEVICE_ADDRESS == aci_evt->params.cmd_rsp.cmd_opcode we retrieve device address from the hardware into a global variable and set addrRetrieved = true
6) once step 4 reaches addrRetrieved=true, we use the device address global variable to create a name based on the device address, then set the advertised name via
 lib_aci_set_local_data(&aci_state, PIPE_GAP_DEVICE_NAME_SET,
                           (unsigned char *)&deviceName, strlen(deviceName));
(see https://devzone.nordicsemi.com/question/6108/change-the-peripheral-name-over-air-through-iphone/ and https://forums.adafruit.com/viewtopic.php?f=19&p=288990)

I guess maybe the device name can be updated at runtime via that PIPE_GAP_DEVICE_NAME_SET call, although that constant is defined in services.h generated by nRF Go.

dakoner, 2016-01-17:
I confirmed: calling begin() twice definitely seems to hang the sketch.

linuxdevel, 2016-01-18:
I use the library and do change both devicename and localname on the fly. But, I delete the current BLEPeripheral class before instanciating it again and setting new names/uuid before calling begin. Seems to work good for my purpose.

dakoner, 2016-01-18:
Thanks, linuxdevel.  That technique worked!  I don't think I would have thought of that.  

I'm going to close this as sandeep said he didn't anticipate this use case, and I have a workaround that works fine.

closed 2016-01-18 by dakoner

sandeepmistry, 2016-01-23:
@dakoner 

> I confirmed: calling begin() twice definitely seems to hang the sketch.

Right, based on @linuxdevel's trick, it's probably running out of RAM, because things are not cleaned up.

I've opened #52 to add a new ```.end()``` API to clean things up, for a nicer way of handling this instead of calling ```delete```

jaymishra01, 2016-05-18:
can you please share your code for changing ble name by the app.