sandeepmistry/arduino-BLEPeripheral #49: Bonding with multiple centrals (specifically iOS devices)

davidlawson, 2016-01-18:
I'm trying to work out how bonding with multiple centrals would be achieved.

My current understanding is that you would have to swap out the bonding data based on which central was trying to connect (I think by calling `lib_aci_write_dynamic_data`). I'm guessing this can be based on the central's address, and can be done after the connection event.

Unfortunately it looks like iOS devices use Random Private Resolvable addresses (involving an Identity Resolution Key). I can't see anything in the repository that resolves the addresses, and I'm not even sure if it's a part of the supported BLE 4.0 specification (I'm using the nRF8001).

Does anyone if I'm on the right track, or know how I can uniquely identify iOS devices for bonding purposes?

sandeepmistry, 2016-01-23:
@davidlawson from my understanding ```lib_ai_read_dynamic_data``` and ```lib_ai_write_dynamic_data``` can only be used when the nRF8001 is in a disconnected state.

@daviddedwin do you have any suggestions?

sandeepmistry, 2016-01-30:
I'm closing this for now, as it's not specifically related to this library, and lack of discussion.

closed 2016-01-30 by sandeepmistry