sandeepmistry/arduino-BLEPeripheral #50: Redbearlab BLE Nano comptatibility

wojciiii, 2016-01-18:
I have a BLE nano board (http://redbearlab.com/blenano/, rev. 1.5). I tried running the same HID keyboard example on it as on an Arduino Uno with a BLE shield. I'm not able to connect to the Nano device. 

Is there any kind of setup that I need to do for this device? Set any of the BLE_REQ/BLE_RDY/BLE_RST - I got the impression that I don't have to set those when using this board.

The examples from https://github.com/RedBearLab/nRF51822-Arduino appar to work. I tried to program the HR example which works out of the box (I can connect to the device from a HR app on my phone).

timotto, 2016-01-21:
I am successfully using the RedBearLabs nRF51822 with Android but I am also having trouble when the Sketch uses the BondStore which HID Peripherals always do. The Android (Nexus 5X, Android 6) starts the pairing but times out after some time. This is the output of the NRF_51822_DEBUG enabled HID_keyboard Sketch:

> version = 8 89 103
> Start advertisement
> BLE HID Keyboard
> Evt Connected 74:bd:ee:d0:0d:e3
> Connected to central: 74:bd:ee:d0:0d:e3
> Evt Sec Params Request 1 1 4 0 0 16
> Evt Conn Param Update 0x6 0x6 0x0 0x7D0
> (~30 seconds pause)
> Evt Conn Param Update 0x1E 0x1E 0x0 0x7D0
> (a few seconds)
> Evt Disconnected
> Start advertisement
> Disconnected from central: 74:bd:ee:d0:0d:e3

Also I can run the Sketch only once, the call 'bleHIDPeripheral.clearBondStoreData();' seems to erase some application code instead of the BondStore data. If I remove it I can run the Sketch again but pairing is rejected right away.

I flashed mk20 & bootloader hex files of the S130 branch of RedBearLab/nRF51822-Arduino@b24f375b35430137c578c4ccacd328ac33e9bce5 My Arduino IDE is 1.6.9 with the 1.0.5 version of the RedBearLab nRF51822 Boards package.

sandeepmistry, 2016-01-23:
@wojciiii 

> I have a BLE nano board (http://redbearlab.com/blenano/, rev. 1.5). I tried running the same HID keyboard example on it as on an Arduino Uno with a BLE shield. I'm not able to connect to the Nano device.

Do the other non-HID examples work?

> Is there any kind of setup that I need to do for this device? Set any of the BLE_REQ/BLE_RDY/BLE_RST - I got the impression that I don't have to set those when using this board.

Correct, those are not used.

sandeepmistry, 2016-01-23:
@timotto please open a new issue.

sandeepmistry, 2016-02-07:
Closing due to lack of activity, please re-open if needed with the request info from https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/50#issuecomment-174194140.

closed 2016-02-07 by sandeepmistry

cheena19, 2017-05-22:
sir the eddystone i made in the same way is detected sometimes and in some apps and sometimes not detected . what can be the reason and how can i fix this issue?