sandeepmistry/arduino-BLEPeripheral #51: Cannot change the timing of a pipe (rf_interval) via BLEPeripheral API

dakoner, 2016-01-21:
We are tuning the timing of a pipe to increase message rate (it's a notification and we want to send ~20+ notifications a second).  

our previous code, where we used the nRF SDK API directly, had this:
    case ACI_EVT_PIPE_STATUS: {
        if (lib_aci_is_pipe_available(&aci_state, PIPE_VALUE_TX) && !timing_change_done) {
          lib_aci_change_timing(6, 6, 0, 600); // Low-latency parameters
        timing_change_done = true;
      }

which is based on the nRF SDK API example:
https://github.com/NordicSemiconductor/ble-sdk-arduino/blob/master/libraries/BLE/examples/ble_bandwidth_test/ble_bandwidth_test.ino#L332

It looks like the BLEPeripheral/BLEDevice classes don't support changing the timing (I grepped the code for calls to lib_aci_change_timing, and didn't see any).

sandeepmistry, 2016-01-23:
@dakoner 

> It looks like the BLEPeripheral/BLEDevice classes don't support changing the timing (I grepped the code for calls to lib_aci_change_timing, and didn't see any).

Correct, there is no API for this.

An API needs to be added to support this, and then implemented for both the nRF8001 and nRF51822.

Note: there is nothing stopping you from calling ```lib_aci_change_timing``` directly now, until this API is added.

sandeepmistry, 2016-03-05:
A new ```BLEPeripheral::setConnectionInterval(min, max)``` API has been added in ae5148ca2f9f9fec281f17ab49922550eba353a1. If called, it will automatically call ```lib_ai_change_timing``` with appropriate values when connection parameters can be updated after a new connection.

closed 2016-03-05 by sandeepmistry