sandeepmistry/arduino-BLEPeripheral #53: RedBearLab Blend Micro (V1.0) & Android compatibility

nicologhielmetti, 2016-01-26:
I'm try to connect the module Blend Micro to my Nexus 5 with Android 6.0.1 and the pairing seems to work: using the HID examples (which are the ones I am interested in), the output on the serial monitor seems correct but it does not perform any "function" on my smartphone. I tried the HID_test.ino sketch, with all the 4 methods of the object bleMouse, bleKeyboard, bleMultimedia, bleSystemControl uncommented to be sure that any class works with my phone. Any suggestion?

Thanks.

P.S.: I set the correct pin 
#define BLE_REQ   6
#define BLE_RDY   7
#define BLE_RST   4
.

sandeepmistry, 2016-01-30:
@nicologhielmetti you'll need to set the HID report offset to 1: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/HID/HID_volume/HID_volume.ino#L46-L48

As Android doesn't not like starting with a report offset of 0. However, this is fine with OS X and iOS.

I'm closing for now, let me know if this works for you.

closed 2016-01-30 by sandeepmistry

nicologhielmetti, 2016-01-31:
Ok, thanks @sandeepmistry, now it works fine! I think it's better to write this suggestion in the first-page, near the settings of pins (but it's only my opinion).

sandeepmistry, 2016-02-07:
@nicologhielmetti great!

Would you be able to submit a pull request to the change you suggested (for all HID examples)?