sandeepmistry/arduino-BLEPeripheral #54: Make connection interval configurable

sandeepmistry, 2016-02-02:
As discussed in https://github.com/01org/corelibs-arduino101/issues/95.

cc/ @soundanalogous

sandeepmistry, 2016-03-05:
Added added in https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/ae5148ca2f9f9fec281f17ab49922550eba353a1.

closed 2016-03-05 by sandeepmistry

soundanalogous, 2016-03-05:
Thanks for adding this! One suggestion however is to document that the min and max values passed to `setConnectionInterval` are the desired time values in milliseconds / 1.25 so if you want to set min as 7.5ms, you need to pass a value of 6 as the minInterval value.

sandeepmistry, 2016-03-06:
@soundanalogous good suggestion!

I've added in f8f319d05dbe4a7411112fa8a0bb39e6aeced964 and 9149a50005abc902c76daf9b055218346e77a18a.