sandeepmistry/arduino-BLEPeripheral #55: analogWrite crashes RFduino

don, 2016-02-07:
analogWrite for values other than 0x00 and 0xFF crash RFduino.

Example sketch https://github.com/don/solidcon-hands-on-ble/blob/master/arduino/LED_v2/LED_v2.ino

ThomasHeb, 2016-02-10:
Hi Don,

I tested the Sketch on a BLE Nano and it works fine.

analogWrite generates a PWM. This must be supported by hardware (see https://www.arduino.cc/en/Reference/AnalogWrite). analogWrite of 0x00 or 0xFF is similar to digitalWirte. This indicates that analogWrite is not supported on PIN6 of RFduino. Unfortunately the docs of the RFduino are not very clear.

You may test with a simple sketch
#define LED_PIN     6

void setup() {
  Serial.begin(9600);

  // set LED pin to output mode
  pinMode(LED_PIN, OUTPUT);

   Serial.println(F("Ready"));
}

void loop() {
  analogWrite(LED_PIN, 0x45); // value other than 0x00 or 0xFF
  Serial.println(F("Alive"));
  delay(500);
}

don, 2016-02-10:
@ThomasHeb analogWrite works with pin 6 when using the RFduino library. As far as I can tell PWM works on most RFduino pins, but it's not well documented.

ThomasHeb, 2016-02-10:
Hi Don,

I went through the RFduino/cores/arduino/wiring_analog.c
the lib looks different to the one installed as standard. so I think you should integrate the rfduino one.

RFduino:
void analogWrite(uint32_t ulPin, uint32_t ulValue) {

  if (ulValue == 0)
  {
    digitalWrite(ulPin, LOW);
    return;
  }

  if (ulValue == 255)
  {
    digitalWrite(ulPin, HIGH);
    return;
  }

	if (Timer1_Compare_Unit_Occupied_by_Pin[0] == ulPin)
	{
		update_PWM_value(ulPin, ulValue, 0);
	}
	else if (Timer1_Compare_Unit_Occupied_by_Pin[1] == ulPin)
	{

Arduino:
void analogWrite(uint8_t pin, int val)
{
	// We need to make sure the PWM output is enabled for those pins
	// that support it, as we turn it off when digitally reading or
	// writing with them.  Also, make sure the pin is in output mode
	// for consistenty with Wiring, which doesn't require a pinMode
	// call for the analog output pins.
	pinMode(pin, OUTPUT);
	if (val == 0)
	{
		digitalWrite(pin, LOW);
	}
	else if (val == 255)
	{
		digitalWrite(pin, HIGH);
	}
	else
	{
		switch(digitalPinToTimer(pin))
		{
			// XXX fix needed for atmega8
			#if defined(TCCR0) && defined(COM00) && !defined(__AVR_ATmega8__)
			case TIMER0A:
				// connect pwm to pin on timer 0
				sbi(TCCR0, COM00);
				OCR0 = val; // set pwm duty
				break;
			#endif

			#if defined(TCCR0A) && defined(COM0A1)
			case TIMER0A:
				// connect pwm to pin on timer 0, channel A
				sbi(TCCR0A, COM0A1);
				OCR0A = val; // set pwm duty
				break;
			#endif

sandeepmistry, 2016-02-13:
@don I've push https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/49fb39f2ce43a551c1f9f9937aa05fc2617d48d2 which sets ```RFduinoBLE_enabled``` to 1 on RFduino inside ```blePeripheral.begin()```. It seems to fixes things for me can you please try it out?

A side note, it really sucks that parts of the RFduino Arduino core are closed source. I found the var by looking around in [variants/RFduino/libRFduino.h](https://github.com/RFduino/RFduino/blob/4e9e0aef93ab954724ab8241edc615313c8c3a1e/variants/RFduino/libRFduino.h#L22-L25). The closed sourced [variants/RFduino/libRFduinoBLE.a](https://github.com/RFduino/RFduino/blob/4e9e0aef93ab954724ab8241edc615313c8c3a1e/variants/RFduino/libRFduinoBLE.a), must be setting it to one as well, and then some "magic" happens ...

closed 2016-02-13 by sandeepmistry

don, 2016-02-13:
@ThomasHeb thanks for looking into this.

@sandeepmistry your fix works great. Thanks!