sandeepmistry/arduino-BLEPeripheral #56: Does not work with SD library

szotyi007, 2016-02-20:
The 0.1.7 version does not work with SD library.  I think some kind of SPI setting is the problem, because if I first setup the ble, then the sd then it is not connecting, but if I set first the sd and then the ble it works. And after a while I do some sd functions, and than it is again broken.

szotyi007, 2016-02-20:
I have got a little closer. I have pulled up chip select pin with (digitalWrite(BLE_REQ_PIN), HIGH) before every SD function, and it looks okay now. I guess in the library it is left out.

sandeepmistry, 2016-02-21:
@szotyi007 please share your full sketch and pins used etc.

szotyi007, 2016-02-21:
please find below my code (i have made it simple to point the problem):
`#include <SPI.h>
#include <SD.h>
#include <stdint.h>
#include <BLEPeripheral.h>

#define BLE_REQ   6
#define BLE_RDY   5
#define BLE_RST   4

BLEPeripheral blePeripheral = BLEPeripheral(BLE_REQ, BLE_RDY, BLE_RST);
BLEService uartService = BLEService("713d0000503e4c75ba943148f18d941e");
BLECharacteristic txCharacteristic = BLECharacteristic("713d0002503e4c75ba943148f18d941e", BLENotify, 20);
BLECharacteristic rxCharacteristic = BLECharacteristic("713d0003503e4c75ba943148f18d941e", BLEWriteWithoutResponse, 20);
BLECentral central = blePeripheral.central();

boolean display_connected = false;

void setup() {
 setup_ble();

 // digitalWrite(6, HIGH); must be used for working..
 if (!SD.begin(8)) {
    Serial.println("initialization failed!");
 }
 File file = SD.open("/");
 file.close();
 
}

void loop() {
 if (display_connected == false) {
    central = blePeripheral.central();
    if (central) {
      display_connected = true;
      Serial.print("Connected to central: ");
      Serial.println(central.address());
    }
  }
  if (display_connected && central.connected()) {
    if (rxCharacteristic.written()) {
      unsigned char len = rxCharacteristic.valueLength();
      const unsigned char *val = rxCharacteristic.value();
      unsigned char i = 0;
      while (i < len) {
        Serial.write(val[i++]);
      }
    }

    if (txCharacteristic.canNotify()) {
      txCharacteristic.setValue((const unsigned char *)"12345", 21);
    }
  }

  if (display_connected == true && central.connected() == false ) {
    display_connected = false;
    Serial.print("Disconnected from central: ");
    Serial.println(central.address());
  }

}

void setup_ble() {
  blePeripheral.setLocalName("customBle");
  blePeripheral.setAdvertisedServiceUuid(uartService.uuid());

  blePeripheral.addAttribute(uartService);
  blePeripheral.addAttribute(rxCharacteristic);
  blePeripheral.addAttribute(txCharacteristic);

  blePeripheral.begin();

  Serial.println(F("BLE UART Peripheral"));
}
`

sandeepmistry, 2016-02-21:
@szotyi007 what hardware are you using?

szotyi007, 2016-02-21:
Arduino Due, with MOLEX 502774-0891 SD socket.

szotyi007, 2016-02-22:
Have you find the cause of the bug? Thank you!

sandeepmistry, 2016-03-06:
@szotyi007 I haven't had time to set this up locally, it would be much easier if you could find a way to reproduce without needing an SD card shield.

Could you please try using a different pin instead of 4 (or 10) for reset?

The lower level nRF8001 SDK already pulls REQ high after each SPI transaction:  https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/utility/hal_aci_tl.cpp#L209-L212

I'm wondering if something in the SPI or SD card library is changing pin 4 or 6, and possibly reseting the nRF8001.

szotyi007, 2016-03-07:
Hi!

Thanks for your email.

I have just left it, i have changed the module to the ble nano..

Thanks again!
Zsolt

sandeepmistry, 2016-03-13:
@szotyi007 ok, I'll close this for now. We can re-open if you or someone else has time to investigate my suggestion on the Due.

closed 2016-03-13 by sandeepmistry

szotyi007, 2016-03-13:
Ok, no problem. Thanks again!

JJANANIPRIYA, 2017-06-27:
@szotyi007 
Hi, I'm using Redbearlabs BLE Nano. The code which you have shared with SD library is not compiling for BLE Nano. any idea? or do have any other code which library compatible with BLE nano library? I want to save data from BLE Nano to SD card in my application.

szotyi007, 2017-06-27:
Hi,
I have posted the code, because it was not working properly :) At the time I was using the ble shield, and it compiled.
I have tried now with nano, I also get a lot of error by compiling. I am not sure, how much time to fix everything. What I do is that I use the nano only for "gateway" between my ipad and the "main processor". I use a teensy 3.6 with built-in micro sd card slot and communicate with ble nano over general ttl serial. If the nano receives data from ipad (via bluetooth), it forwards the data to teensy (over serial) and do nothing else, and I actually parse data on the teensy. If I need to send information to ipad then I send it from teensy to nano (over serial) and the nano "forwards" to ipad.
A simple version of my code for ble nano:
```
#include <stdint.h>
#include <SPI.h>
#include <BLEPeripheral.h>

#define BLE_REQ   9
#define BLE_RDY   8
#define BLE_RST   10

BLEPeripheral blePeripheral = BLEPeripheral(BLE_REQ, BLE_RDY, BLE_RST);
BLEService uartService = BLEService("713d0000503e4c75ba943148f18d941e");
BLECharacteristic txCharacteristic = BLECharacteristic("713d0002503e4c75ba943148f18d941e", BLENotify, 20);
BLECharacteristic rxCharacteristic = BLECharacteristic("713d0003503e4c75ba943148f18d941e", BLEWriteWithoutResponse, 20);

BLECentral central = blePeripheral.central();

char buffer[21];
int buffer_counter = 0;
unsigned long buffer_lastmillis = 0;

boolean display_connected = false;

void setup() {
  Serial.begin(38400); // does not work properly with higher baundrate
  pinMode(13, OUTPUT);

  blePeripheral.setLocalName("testName");
  blePeripheral.setAdvertisedServiceUuid(uartService.uuid());

  blePeripheral.addAttribute(uartService);
  blePeripheral.addAttribute(rxCharacteristic);
  blePeripheral.addAttribute(txCharacteristic);

  blePeripheral.begin();

  Serial.println("GNSS BlE starts");

}
unsigned long counter = 0;
unsigned long lastmillis = 0;

void loop() {
  //Serial.println("lol");
  if (millis() - lastmillis > 1000) {
    digitalWrite(13, !digitalRead(13));
    lastmillis = millis();
  }
  central = blePeripheral.central();
  if (display_connected == false) {
    if (central) {
      display_connected = true;
      Serial.print("Connected to central: ");
      Serial.println(central.address());
    }
  }
  if (display_connected && central.connected()) {
    if (rxCharacteristic.written()) {
      unsigned char len = rxCharacteristic.valueLength();
      const unsigned char *val = rxCharacteristic.value();

      unsigned char i = 0;
      while (i < len) {Serial.write(val[i++]);}
      //Serial.println("");
    }

    if (txCharacteristic.canNotify()) {
      // txCharacteristic.setValue((const unsigned char *)"test", 1);
    }
  }

  if (display_connected == true && central.connected() == false ) {
    display_connected = false;
    Serial.print("Disconnected from central: ");
    Serial.println(central.address());
  }

  read_serial();
}

void read_serial() {
  if (Serial.available()) {

    for ( int i = 0; i < sizeof(buffer); i++) buffer[i] = (char)0;
    buffer_counter = 0;

    unsigned long buffer_lastmillis = millis();
    while (Serial.available() || millis() - buffer_lastmillis < 20) {
      if (Serial.available()) {
        char readed_char = Serial.read();
       // Serial.write(readed_char);
        buffer[buffer_counter] = readed_char;
        buffer_counter++;
        buffer_lastmillis = millis();

        if (buffer_counter == 20) {
          while (!txCharacteristic.canNotify()) { // must be modified for timout 
            blePeripheral.connected();
            //delay(20);
          }
          txCharacteristic.setValue((const unsigned char *)buffer, 20);
          for ( int i = 0; i < sizeof(buffer); i++) buffer[i] = (char)0;
          buffer_counter = 0;
        }
      }
    }
    if (buffer_counter > 0) {
      while (!txCharacteristic.canNotify()) { // must be modified for timout
        blePeripheral.connected();
      }
      
      txCharacteristic.setValue((const unsigned char *)buffer, 20);
    }
  }
}
```