sandeepmistry/arduino-BLEPeripheral #57: hi if i use rfduino, how do i set this function "BLEHIDPeripheral bleHIDPeripheral = BLEHIDPeripheral(BLE_REQ, BLE_RDY, BLE_RST);"

tdzuc, 2016-03-19:
hi 
Thank you for your wonderful library.
I use rfduino borad, but i don.t konw how to set this function " BLEHIDPeripheral(BLE_REQ, BLE_RDY, BLE_RST)". Because the Pins for BLE_REQ, BLE_RDY, BLE_RST are not used, the question is 
how do i set this function "BLEHIDPeripheral bleHIDPeripheral = BLEHIDPeripheral(BLE_REQ, BLE_RDY, BLE_RST);"

THANKS you very much.

tdzuc, 2016-03-19:
Sorry, i found you write the code in BLEPeripheral.h.
like below
#if defined(NRF51) || defined(__RFduino__)
  #define BLE_DEFAULT_REQ   -1
  #define BLE_DEFAULT_RDY   -1
  #define BLE_DEFAULT_RST   -1
So i can set any number in Pins
#define BLE_REQ   9
#define BLE_RDY   8
#define BLE_RST   4

Because it will set it automatically. THANKS

sandeepmistry, 2016-03-20:
I've updated the table to indicated this clearer: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/README.md#pinouts

closed 2016-03-20 by sandeepmistry