sandeepmistry/arduino-BLEPeripheral #59: hi i  use the example HID_joystick_mouse in BLEPeripheral library and i test it in cell phone. Althoung it recevied the data from BLE but it seems not control the  cursor.

tdzuc, 2016-03-21:
hi i  use the example HID_joystick_mouse in BLEPeripheral library and i test it in cell phone. Actually the cell phone recevied the data from BLE like 00 0f ff 01. 
because i  set :       
      int x=15;
      int y=255;
      if (x || y) {
        bleMouse.move(x, y,1);
        Serial.println(x);
        Serial.println(y);
      } 
but it seems not control the  cursor.
i modify the code as below. I suppose it should control the cursor and i can select the item in the cell phone, like the ipeda bluetooth controller, shown as below. May you have any idea about that?
Or i need to set some configuration in GATT.
https://www.youtube.com/watch?v=Ua6MDKx2xwg

void loop() {
  BLECentral central = bleHIDPeripheral.central();

  if (central) {
    // central connected to peripheral
    Serial.print(F("Connected to central: "));
    Serial.println(central.address());

    while (central.connected()) {
      Serial.println(central.connected())
      int x=15;
      int y=255;
      if (x || y) {
        bleMouse.move(x, y,1);
        Serial.println(x);
        Serial.println(y);
      }
    }

    // central disconnected
    Serial.print(F("Disconnected from central: "));
    Serial.println(central.address());
  }
}

sandeepmistry, 2016-03-27:
@tdzuc are you using a Android device?

If so, you'll probably need the following line in a similar place in the sketch: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/HID/HID_volume/HID_volume.ino#L47

Android does not support the default report id offset of 0 ...

tdzuc, 2016-03-28:
wa,it works after i add 
bleHIDPeripheral.setReportIdOffset(1);
i use RFDUINO.

thank you very much.

sandeepmistry, 2016-04-03:
@tdzuc great!

closed 2016-04-03 by sandeepmistry