sandeepmistry/arduino-BLEPeripheral #60: hi i want write a gamepag HID descriptor i have some question about the function

tdzuc, 2016-03-29:
hi i want write a gamepag HID descriptor i have some question about the function

BLEMouse::BLEMouse() :
  BLEHID(descriptorValue, sizeof(descriptorValue), 11),
  _reportCharacteristic("2a4d", BLERead | BLENotify, 4),
  _reportReferenceDescriptor(BLEHIDDescriptorTypeInput),
  _button(0)
{
}

1. how do you decide the number 11? it seems like reportIdOffset, but how do you decide to set 11?
BLEHID::BLEHID(const unsigned char* descriptor, unsigned char descriptorLength, unsigned char reportIdOffset) :
  _reportId(0),
  _descriptor(descriptor),
  _descriptorLength(descriptorLength),
  _reportIdOffset(reportIdOffset)
{
  _numHids++;
}

2._reportCharacteristic("2a4d", BLERead | BLENotify, 4),
the number represent the send value is 4 bytes
but the MOUSE HID DESCRIPTOR only set 
3 button  1bytes
1 X         1bytes
1Y          1bytes
the last byte is wheel but i didn't see it in mouse HID descriptor

3.i will use the descriptor like below to control the gamepad.

how do i set the number 11 and 4 as below
  BLEHID(descriptorValue, sizeof(descriptorValue), 11),
  _reportCharacteristic("2a4d", BLERead | BLENotify, 4),

0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
0x85, 0x07,        //   Report ID (7)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x09, 0x32,        //     Usage (Z)
0x09, 0x35,        //     Usage (Rz)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x04,        //     Report Count (4)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection

Thank you for always answer my question so quickly.

tdzuc, 2016-03-30:
Hi ,
below is my code for gamepad. the android received the data but it still seems not control the phone.
i push the joystick up, i think the selection of the item should be rotated up, but it seems not anything happen. i already add " bleHIDPeripheral.setReportIdOffset(1);"   

**my BLEGamepad.cpp**

#include "BLEGamepad.h"

static const PROGMEM unsigned char descriptorValue[] = {
  // From: https://github.com/adafruit/Adafruit-Trinket-USB/blob/master/TrinketHidCombo/TrinketHidComboC.c
  //       permission to use under MIT license by @ladyada (https://github.com/adafruit/Adafruit-Trinket-USB/issues/10)

0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
0x85, 0x07,        //   Report ID (7)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x09, 0x32,        //     Usage (Z)
0x09, 0x35,        //     Usage (Rz)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x04,        //     Report Count (4)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection

0x09, 0x39,        //   Usage (Hat switch)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x07,        //   Logical Maximum (7)
0x35, 0x00,        //   Physical Minimum (0)
0x46, 0x3B, 0x01,  //   Physical Maximum (315)
0x65, 0x14,        //   Unit (System: English Rotation, Length: Centimeter)
0x75, 0x04,        //   Report Size (4)
0x95, 0x01,        //   Report Count (1)
0x81, 0x42,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,Null State)
0x75, 0x04,        //   Report Size (4)
0x95, 0x01,        //   Report Count (1)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x09,        //   Usage Page (Button)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x0F,        //   Usage Maximum (0x0F)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x10,        //   Report Count (16)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0              // End Collection

};

BLEGamepad::BLEGamepad() :
  BLEHID(descriptorValue, sizeof(descriptorValue), 7),
  _reportCharacteristic("2a4d", BLERead | BLENotify, 7),
  _reportReferenceDescriptor(BLEHIDDescriptorTypeInput)
{
}
void BLEGamepad::move(signed char x, signed char y, signed char z, signed char Rz) {
  unsigned char gamepadMove[7]= { 0x00, 0x00, 0x00, 0x00,0x00,0x00,0x00};

  // send key code
  gamepadMove[0] = x;
  gamepadMove[1] = y;
  gamepadMove[2] = z;
  gamepadMove[3] = Rz;
  this->sendData(this->_reportCharacteristic, gamepadMove, sizeof(gamepadMove));
}

void BLEGamepad::setReportId(unsigned char reportId) {
  BLEHID::setReportId(reportId);

  this->_reportReferenceDescriptor.setReportId(reportId);
}

unsigned char BLEGamepad::numAttributes() {
  return 2;
}

BLELocalAttribute** BLEGamepad::attributes() {
  static BLELocalAttribute* attributes[2];

  attributes[0] = &this->_reportCharacteristic;
  attributes[1] = &this->_reportReferenceDescriptor;

  return attributes;
}

**My BLEGamepad.h**

#ifndef _BLE_GAMEPAD_H_
#define _BLE_GAMEPAD_H_

#include "Arduino.h"

#include "BLECharacteristic.h"
#include "BLEHIDReportReferenceDescriptor.h"
#include "BLEHID.h"

// From: https://github.com/adafruit/Adafruit-Trinket-USB/blob/master/TrinketHidCombo/TrinketHidCombo.h
//       permission to use under MIT license by @ladyada (https://github.com/adafruit/Adafruit-Trinket-USB/issues/10)

class BLEGamepad : public BLEHID
{
  public:
    BLEGamepad();

    void move(signed char x, signed char y, signed char z, signed char Rz);

  protected:
    virtual void setReportId(unsigned char reportId);
    virtual unsigned char numAttributes();
    virtual BLELocalAttribute** attributes();

  private:
    BLECharacteristic                 _reportCharacteristic;
    BLEHIDReportReferenceDescriptor   _reportReferenceDescriptor;
};

#endif

**My Arduino code**

#include <SPI.h>
#include <BLEHIDPeripheral.h>
#include <BLEGamepad.h>

// define pins (varies per shield/board)
#define BLE_REQ   -1
#define BLE_RDY   -1
#define BLE_RST   -1

// create peripheral instance, see pinouts above
BLEHIDPeripheral bleHIDPeripheral = BLEHIDPeripheral(BLE_REQ, BLE_RDY, BLE_RST);
BLEGamepad bleGamepad;

void setup() {
  Serial.begin(9600);
#if defined (__AVR_ATmega32U4__)
  while(!Serial);
#endif
  bleHIDPeripheral.clearBondStoreData();
  bleHIDPeripheral.setReportIdOffset(1);   
  bleHIDPeripheral.setLocalName("HID Gamepad");
  bleHIDPeripheral.addHID(bleGamepad);

  bleHIDPeripheral.begin();

  Serial.println(F("BLE HID Gamepad"));
  BLECentral central = bleHIDPeripheral.central();
}

void loop() {
  BLECentral central = bleHIDPeripheral.central();

  if (central) {
    // central connected to peripheral
    Serial.print(F("Connected to central: "));
    Serial.println(central.address());

    while (central.connected()) {
      
    int x=0;
    int y=14;
    int z=0;
    int Rz=0;
       bleGamepad.move(x, y, z, Rz);
    }

    // central disconnected
    Serial.print(F("Disconnected from central: "));
    Serial.println(central.address());
  }
}

sandeepmistry, 2016-04-03:
@tdzuc 

> 1.how do you decide the number 11? it seems like reportIdOffset, but how do you decide to set 11? 

This is the byte offset of the report offset in the ```descriptorValue```.

See: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/BLEMouse.cpp#L12

> 2._reportCharacteristic("2a4d", BLERead | BLENotify, 4), the number represent the send value is 4 bytes but the MOUSE HID DESCRIPTOR only set  3 button 1bytes 1 X 1bytes 1Y 1bytes
the last byte is wheel but i didn't see it in mouse HID descriptor

Maybe wheel is not supported then?

> 3.i will use the descriptor like below to control the gamepad.

I think you need to use 7 for the report ID offset.

P.S.: please use markdown formatting in your comments, it makes things easier to read.

sandeepmistry, 2016-05-09:
I think this has been answered, so closing for now.

closed 2016-05-09 by sandeepmistry