sandeepmistry/arduino-BLEPeripheral #61: Blend micro's default service can't be override by my code using BLEPeripheral library

jimmyntu, 2016-03-29:
Hi, my redbear labs Blend Micro worked before with the new service I registered using BLEPeripheral library. Your library really helped a lot! Thanks! But after some of the update, I'm not sure what is changed. The new service and characteristics are not shown in my Blend Micro. It always show the default Blend TX/RX service '713d0000-503e-4c75-ba94-3148f18d941e'. My created service is never shown. The board only shows the service id and device name in advertisement data. But the real service is still blend micro's default one. 

Could you help me to identify what's the problem? Thanks a lot in advance!

Arduino Version 1.6.8 (Tried in 1.6.6, same problem)
RedbearLab AVR Boards version 1.0.2 (for blend micro)
BLEPeripheral version 0.1.8
RBL_nRF8001 version 1.0.0

Here is my code
`
// Import libraries (BLEPeripheral depends on SPI)
#include <SPI.h>
#include <BLEPeripheral.h>

// define pins (varies per shield/board, UNUSED for nRF51822)
#define BLE_REQ     6
#define BLE_RDY     7
#define BLE_RST     4

// LED and button pin
#define SWITCH_PIN     3

// create peripheral instance, see pinouts above
BLEPeripheral            blePeripheral        = BLEPeripheral(BLE_REQ, BLE_RDY, BLE_RST);

// create service
BLEService               ledService           = BLEService("ffe0");

// create device id characteristic
BLECharacteristic deviceIdChar = BLECharacteristic("ffe1", BLERead | BLEWrite, 20);

void setup() {
  Serial.begin(115200);
#if defined (__AVR_ATmega32U4__)
  delay(5000);  //5 seconds delay for enabling to see the start up comments on the serial board
#endif

  // set LED pin to output mode, button pin to input mode
  pinMode(SWITCH_PIN, OUTPUT);
  digitalWrite(SWITCH_PIN, LOW);

  // set advertised local name and service UUID
  blePeripheral.setDeviceName("LED");
  blePeripheral.setLocalName("LED");
  blePeripheral.setAdvertisedServiceUuid(ledService.uuid());

  // add service and characteristics
  blePeripheral.addAttribute(ledService);
  blePeripheral.addAttribute(deviceIdChar);

  // begin initialization
  blePeripheral.begin();

  Serial.println(F("BLE LED Peripheral"));
}

void loop() {
  // poll peripheral
  blePeripheral.poll();

  BLECentral central = blePeripheral.central();

  if (central)
  {      
      // has the value changed since the last read
      if (deviceIdChar.written()) {
        unsigned char len = deviceIdChar.valueLength();
        const unsigned char *data = deviceIdChar.value();
        Serial.println(len, DEC);
        deviceIdChar.setValue(data, (int)len);
      }
  }
}
`

jimmyntu, 2016-03-30:
Sorry. My bad. I think the code works. The problem is I need to unplug the board and plug it again to let the code works. Sorry for the wrong message.

closed 2016-03-30 by jimmyntu

sandeepmistry, 2016-04-03:
@jimmyntu be sure to update the ```BLE_*``` pins to match the jumpers on your Blend boards.

Also, if you don't have a reset jumper plugged in you might have to power cycle the board.