sandeepmistry/arduino-BLEPeripheral #64: HID not working after reconnect

teosnare, 2016-04-04:
Hi
I'm experimenting with HID (HID_volume) on RFduino.
At the moment I'm having an issue on device reconnect.
The first time I connect the RFduino with my smartphone, the volume function works correctly.
If the RFduino is already paired with my phone and I reconnect it, the phone doesn't receive the volume command.
I'm trying on Samsung J5 and also on a Asus Laptop.
I activated the NRF_51822_DEBUG flag and what I see is this:

```
Starting...
BLE HID Peripheral - clearing bond data
version = 6 89 46
Start advertisement
BLE HID Volume Knob
Evt Connected dc:85:de:e5:08:30
Connected to central: dc:85:de:e5:08:30
Evt Sec Params Request 0 1 1 4 0 0 16
Evt Conn Sec Update 1 2 16
Evt Auth Status
0
Storing bond data
Evt Conn Param Update 0xC 0xC 0x0 0x3C0
Evt Sys Attr Missing 0
Evt Write, handle = 11
02 00 
Evt Write, handle = 21
01 00 
Evt Conn Param Update 0x30 0x30 0x0 0x3C0
Evt Conn Param Update 0x8 0x8 0x0 0xC8

#after sending HID key
Evt TX complete 1
Evt TX complete 1

#Turning off and on the BT on the host:

Evt Disconnected
Start advertisement
Disconnected from central: dc:85:de:e5:08:30
Evt Connected dc:85:de:e5:08:30
Connected to central: dc:85:de:e5:08:30
Evt Sec Info Request 16189 1 0 0
Evt Conn Sec Update 1 2 16

#trying to send HID keys but no effect

#Resetting RFduino without clearing bond data:

Starting...
version = 6 89 46
Restoring bond data
Start advertisement
BLE HID Volume Knob
Evt Connected dc:85:de:e5:08:30
Connected to central: dc:85:de:e5:08:30
Evt Sec Info Request 16189 1 0 0
Evt Conn Sec Update 1 2 16

#again trying to send HID keys but no effect
```

I'm digging deeper with some more debug and it seems that a possible problem is that 
localCharacteristicInfo->notifySubscribed
in the method:
bool nRF51822::updateCharacteristicValue(BLECharacteristic& characteristic) 
is FALSE
(row 961 in nRF51822.cpp)

Can you please help me?
Thankyou

Matteo

sandeepmistry, 2016-05-15:
@teosnare how did you get the HID_Volume sketch to build on the RFduino? I am getting build errors with the Encoder library.

I just tried the HID_keyboard example with my iPhone with turning Bluetooth off and on again like your test and it seems fine. Which version of Android is your Samsung J5 using?

teosnare, 2016-05-17:
I'm using RFduino library version 2.3.2 and your BLEPeripheral 0.1.8

sandeepmistry, 2016-06-19:
What about Android version?

sandeepmistry, 2016-08-14:
Closing this due to lack of feedback. Please re-open with requested info.

closed 2016-08-14 by sandeepmistry

cheeta1, 2017-11-19:
hi @teosnare  did you solved this issue? i am having the same issue with my nrf51822 module.

cheeta1, 2017-11-21:
hi @sandeepmistry if you need any debug info then i can provide you. if nrf51822 is connected to your phone and you reset its power then the nrf51822 device reconnects but it does not send any data. we have to disconnect it from iphone or android( i am using Android 8 and 6 both have issues) then we have to forget it and pair it again to work.

cheeta1, 2018-01-06:
hi @teosnare  have you solved the issue if yes please tell me how to solve it?

Eddddddddy, 2020-04-28:
@cheeta1 have you solved this problem? Or some useful information? i really need it.