sandeepmistry/arduino-BLEPeripheral #66: HID output reports? (also multiple report IDs)

RapperDMX, 2016-04-13:
Maybe I've overlooked something obvious, but are there any guides/samples to follow when using HID _**output**_ reports to send data from central to the peripheral (to control LEDs, motors, etc).

I see a reference to it in the keyboard sample in the mentioned LEDs but nothing fleshed out to work from.
( ~/arduino-BLEPeripheral/examples/HID/HID_keyboard/ )

Also, is there a pointer on how to receive  the data from the second or third (etc) HID report?

I ask because my reverse engineering attempts on this gaming controller at the point now where the primary button/joystick inputs in the first report work perfectly, but now I want to (attempt to) send force feedback rumble commands.

Here is the (USB derived) descriptor i'm working with if anyone's curious.

`

0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
	0xA1, 0x02,        //   Collection (Logical)
		0x85, 0x01,        //     Report ID (1)
		0x75, 0x08,        //     Report Size (8)
		0x95, 0x04,        //     Report Count (4)
		0x15, 0x00,        //     Logical Minimum (0)
		0x26, 0xFF, 0x00,  //     Logical Maximum (255)
		0x35, 0x00,        //     Physical Minimum (0)
		0x46, 0xFF, 0x00,  //     Physical Maximum (255)
		0x09, 0x30,        //     Usage (X)
		0x09, 0x31,        //     Usage (Y)
		0x09, 0x32,        //     Usage (Z)
		0x09, 0x35,        //     Usage (Rz)
		0x81, 0x02,        //     **Input** (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
		0x75, 0x04,        //     Report Size (4)
		0x95, 0x01,        //     Report Count (1)
		0x25, 0x07,        //     Logical Maximum (7)
		0x46, 0x3B, 0x01,  //     Physical Maximum (315)
		0x66, 0x14, 0x00,  //     Unit (System: English Rotation, Length: Centimeter)
		0x09, 0x39,        //     Usage (Hat switch)
		0x81, 0x42,        //     **Input** (Data,Var,Abs,No Wrap,Linear,Preferred State,Null State)
		0x66, 0x00, 0x00,  //     Unit (None)
		0x75, 0x01,        //     Report Size (1)
		0x95, 0x0C,        //     Report Count (12)
		0x25, 0x01,        //     Logical Maximum (1)
		0x45, 0x01,        //     Physical Maximum (1)
		0x05, 0x09,        //     Usage Page (Button)
		0x19, 0x01,        //     Usage Minimum (0x01)
		0x29, 0x0C,        //     Usage Maximum (0x0C)
		0x81, 0x02,        //     **Input** (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
		0x95, 0x01,        //     Report Count (1)
		0x75, 0x08,        //     Report Size (8)
		0x06, 0x00, 0xFF,  //     Usage Page (Vendor Defined 0xFF00)
		0x26, 0xFF, 0x00,  //     Logical Maximum (255)
		0x46, 0xFF, 0x00,  //     Physical Maximum (255)
		0x09, 0x00,        //     Usage (0x00)
		0x81, 0x02,        //     **Input** (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
	0xC0,              //   End Collection

	0xA1, 0x02,        //   Collection (Logical)
		0x85, 0x02,        //     Report ID (2)
		0x95, 0x07,        //     Report Count (7)
		0x75, 0x08,        //     Report Size (8)
		0x26, 0xFF, 0x00,  //     Logical Maximum (255)
		0x46, 0xFF, 0x00,  //     Physical Maximum (255)
		0x06, 0x00, 0xFF,  //     Usage Page (Vendor Defined 0xFF00)
		0x09, 0x03,        //     Usage (0x03)
		0x81, 0x02,        //     **Input** (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
	0xC0,              //   End Collection

	0xA1, 0x02,        //   Collection (Logical)
		0x85, 0x03,        //     Report ID (3)
		0x09, 0x04,        //     Usage (0x04)
		0x91, 0x02,        //     **_Output_** (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
	0xC0,              //   End Collection
0xC0,              // End Collection

// 119 bytes

// best guess: USB HID Report Descriptor

`

Thanks for the great work on this library, without it I would have still be stuck fighting the losing battle against nrfgo studio

RapperDMX, 2016-04-13:
As i'm looking deeper into this, i think the key is in the HID object instance itself for example the keyboard

```
BLEKeyboard::BLEKeyboard() :
  BLEHID(descriptorValue, sizeof(descriptorValue), 7),
  _reportCharacteristic("2a4d", BLERead | BLENotify, 8),
  _reportReferenceDescriptor(BLEHIDDescriptorTypeInput)
{
  memset(this->_value, 0, sizeof(this->_value));
}
```

```
enum BLEHIDDescriptorType {
  BLEHIDDescriptorTypeInput = 0x01,        <---------
  BLEHIDDescriptorTypeOutput = 0x02        <---------
};

class BLEHIDReportReferenceDescriptor : public BLEDescriptor
{
  public:
    BLEHIDReportReferenceDescriptor(BLEHIDDescriptorType type);
    virtual ~BLEHIDReportReferenceDescriptor();
    void setReportId(unsigned char reportId);
  private:
    unsigned char         _value[2];
};

```

I'm trying to form a full understanding of this ,but the working theory at the moment is that the HID object (_BLEKeyboard_ here) by default is only able to handle one report, it seems that to get data from (and pehaps to) the other ReportIDs defined in the entire HID Report Descriptor you need additional <**BLEHIDReportReferenceDescriptor**> members with the appropriate <**BLEHIDDescriptorType**> type of input or output specified.  I'll try this theory later and report back my findings.

RapperDMX, 2016-04-26:
Further research into this is seeming to show that it's not possible (perhaps that's why companies like Logitech have a proprietary RF dongle for their rumble gamepads rather than true bluetooth).

I tried spoofing the vid/pid of the logitech F710 so linux would detect it and use the (USB) rumble aware xpad driver, but no luck.

I added another "rumbler" BLEHID device and included it with bleHIDPeripheral.addHID(), the rumbler class defined as
```
    XpadRumble::XpadRumble() :
        BLEHID(descriptorValue, sizeof(descriptorValue), 112),  //112 bytes offset into descriptor
      _reportCharacteristicOutput("2a4d", BLERead | BLENotify | BLEWrite, OUTPUTREPORTLENGTH),
      //_reportCharacteristicOutput("2a4d", BLEWrite, 8), //8 bytes report len
      _reportReferenceDescriptorOutput(BLEHIDDescriptorTypeOutput)
    {
        this->clearOutputReport();
        this->setReportId(3);
    }
```

This unfortunately didn't work either.  Perhaps it is indeed a limitation of bluetooth proto, despite HoGP being described as a wrapper around traditional USB HID.  I can manually through code or nordic tools find and set the data in that output report which i pick up in the rumble report characteristic, but no applications or linux utilities seem to be aware of this output report and it's purpose.

The search continutes...

RapperDMX, 2016-04-26:
The annotated report showing the offsets I used

```
static const PROGMEM unsigned char descriptorValue[] = {

0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)

    0xA1, 0x02,        //   Collection (Logical)
        0x85, 0x01,        //     Report ID (1)                 //offset = 9 bytes
        0x75, 0x08,        //     Report Size (8)
        0x95, 0x04,        //     Report Count (4)
        0x15, 0x00,        //     Logical Minimum (0)
        0x26, 0xFF, 0x00,  //     Logical Maximum (255)
        0x35, 0x00,        //     Physical Minimum (0)
        0x46, 0xFF, 0x00,  //     Physical Maximum (255)
        0x09, 0x30,        //     Usage (X)
        0x09, 0x31,        //     Usage (Y)
        0x09, 0x32,        //     Usage (Z)
        0x09, 0x35,        //     Usage (Rz)
        0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
        0x75, 0x04,        //     Report Size (4)
        0x95, 0x01,        //     Report Count (1)
        0x25, 0x07,        //     Logical Maximum (7)
        0x46, 0x3B, 0x01,  //     Physical Maximum (315)
        0x66, 0x14, 0x00,  //     Unit (System: English Rotation, Length: Centimeter)
        0x09, 0x39,        //     Usage (Hat switch)
        0x81, 0x42,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,Null State)
        0x66, 0x00, 0x00,  //     Unit (None)
        0x75, 0x01,        //     Report Size (1)
            0x95, 0x0C,        //     Report Count (*****)
        0x25, 0x01,        //     Logical Maximum (1)
        0x45, 0x01,        //     Physical Maximum (1)
        0x05, 0x09,        //     Usage Page (Button)
        0x19, 0x01,        //     Usage Minimum (0x01)
            0x29, 0x04,        //     Usage Maximum (******)
        0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
        0x95, 0x01,        //     Report Count (1)
        0x75, 0x08,        //     Report Size (8)
        0x06, 0x00, 0xFF,  //     Usage Page (Vendor Defined 0xFF00)
        0x26, 0xFF, 0x00,  //     Logical Maximum (255)
        0x46, 0xFF, 0x00,  //     Physical Maximum (255)
        0x09, 0x00,        //     Usage (0x00)
        0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,              //   End Collection

    
    0xA1, 0x02,        //   Collection (Logical)
        0x85, 0x02,        //     Report ID (2)                     //offset = 90 bytes
        0x95, 0x07,        //     Report Count (7)
        0x75, 0x08,        //     Report Size (8)
        0x26, 0xFF, 0x00,  //     Logical Maximum (255)
        0x46, 0xFF, 0x00,  //     Physical Maximum (255)
        0x06, 0x00, 0xFF,  //     Usage Page (Vendor Defined 0xFF00)
        0x09, 0x03,        //     Usage (0x03)
        0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,              //   End Collection

    
    0xA1, 0x02,        //   Collection (Logical)
        0x85, 0x03,        //     Report ID (3)                 //offset = 112 bytes
        0x09, 0x04,        //     Usage (0x04)
        0x91, 0x02,        //     Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0xC0,              //   End Collection
    
0xC0,              // End Collection

};

```

sandeepmistry, 2016-05-01:
Hi @RapperDMX,

Unfortunately, I am not a HID report expert, I had used the original ones from https://github.com/adafruit/Adafruit-Trinket-USB.

For multiple reports you probably need to change the return type of ```::numAttributes()``` as well as ```::attributes()```.

RapperDMX, 2016-05-06:
Regrettably, Neither am I, so much of this is iterative experimentation.  I though I had tried addressing the additional attributes before in previous attempts but I will look into it again to make sure I didn't overlook something.

```
unsigned char bleJoyFFB::numAttributes() 
{
  //return 2;
   return 3;
}

BLELocalAttribute** bleJoyFFB::attributes() 
{
  static BLELocalAttribute* attributes[3];

  attributes[0] = &this->_reportCharacteristic;
  attributes[1] = &this->_reportReferenceDescriptor;
  attributes[2] = &this->_reportReferenceDescriptorOutput;   //  <--- here

  return attributes;
}

```

At the moment i'm looking into something like a new BLEHID_MultipeReport class?  One where ::numattributes() is calculated by from some array of "report entries" {reportId,reportByteOffset,reportLength}, same for ::attributes(), hopefully I stumble across something I hook into linux xpad driver with but we'll see.

sandeepmistry, 2016-05-09:
@RapperDMX 

I wonder if you should try to get this working over USB first, using Leonardo/Due/Zero/MKR1000?

sandeepmistry, 2016-06-20:
I'm closing this for now due to lack of activity.

closed 2016-06-20 by sandeepmistry