sandeepmistry/arduino-BLEPeripheral #69: Adafruit Feather 32u4 Bluefruit LE

orcinz, 2016-06-14:
Hi.

I have an Adafruit Feather 32u4 Bluefruit LE use nRF51822 chipset.

Their official software doesnt support to modify HID. They just have mouse and keyboard option.

But i need to use my nRF51822 as gamepad device.
So I tried to upload your code on my Adafruit Feather, but nothing changed. 
It seems like they dont communicate with each other.
Is this maybe because of BLE_REQ, BLE_RDY, BLE_RST keys?

If so, how can I find my Adafruit Feather's pin combination?

Thanks in advance.

zrecore, 2016-06-14:
When you open up an Adafruit Feather M0 Bluefruit LE example (Arduino IDE), it also opens up a config file. Check the values in the config file "BluefruitConfig.h" that is included in each example.

zrecore, 2016-06-14:
Additionally, the example sketches from Adafruit are configured to use hardware SPI near the top of each sketch.

orcinz, 2016-06-14:
Hi. Thank you for your answer @zrecore 
Yes, I have checked the sketch;
https://cdn-learn.adafruit.com/assets/assets/000/030/930/original/adafruit_products_2889_pinout_v1_0.png?1457306365

That shows RST, CS and IRQ pins.

And BluefruitConfig.h has these pins;

#define BLUEFRUIT_SWUART_RXD_PIN       9    // Required for software serial!
#define BLUEFRUIT_SWUART_TXD_PIN       10   // Required for software serial!
#define BLUEFRUIT_UART_CTS_PIN         11   // Required for software serial!
#define BLUEFRUIT_UART_RTS_PIN         -1   // Optional, set to -1 if unused
#define BLUEFRUIT_UART_MODE_PIN        12    // Set to -1 if unused
#define BLUEFRUIT_SPI_CS               8
#define BLUEFRUIT_SPI_IRQ              7
#define BLUEFRUIT_SPI_RST              4    // Optional but recommended, set to -1 if unused

Any opinion if which one is what?.. Or how should I configure?..

zrecore, 2016-06-14:
They can't both act as peripherals and talk to each other (yet). You need
them to talk to a central device.

https://learn.adafruit.com/introduction-to-bluetooth-low-energy/gatt

orcinz, 2016-06-14:
I have read it now. But couldnt be sure to understand exactly. It says, bluetooth device should connect to a centeral device such as mobile phone etc..

So, that means, my adafruit feather and my pc can't connect to each other as a HID gamepad?..

orcinz, 2016-06-14:
Any opinion @zrecore ?

zrecore, 2016-06-14:
Yes, a BLE peripheral can connect to a PC, or a smart device (assuming BLE
support).

The PC or smartphone would act as the central device. The Bluefruit LE
board would act as the peripheral.

Do you have your source code on github? Maybe taking a look at it will help.

orcinz, 2016-06-14:
Well, that is the problem already @zrecore 

Adafruit official library dont let me configure my device as HID gamepad.

` bleHIDPeripheral.setLocalName("HID Keyboard");
  bleHIDPeripheral.addHID(bleKeyboard);

  bleHIDPeripheral.begin();

  Serial.println(F("BLE HID Keyboard"));`

This is the code for HID Keyboard configuration. But there is no such code for HID gamepad.

Thats why I wanted to use arduino-BLEPeripheral library. So that i could connect my adafruit feather as a HID Gamepad..

_"The hardware can do it, but we don't have any existing code that configures the nRF51822 to act like a joystick."_

Thats what support told me when I ask Adafruit forums..

orcinz, 2016-06-14:
Any idea @zrecore ?

zrecore, 2016-06-14:
Most likely, you will need to create an HID Joystick profile (either with
the BLEPeripheral API, or with the Nordic Semiconductor Developers Kit)

Your device must provide the correct descriptor for an HID Joystick.
Perhaps you can research this a bit further.
https://github.com/troystribling/nordic-ble-sdk-examples

Any idea @zrecore <https://github.com/zrecore> ?

orcinz, 2016-06-15:
Tried that too.. Not working..

Do you have any idea about REQ, RDY and RST pins of adafruit feather according to that sketch;

https://cdn-learn.adafruit.com/assets/assets/000/030/930/original/adafruit_products_2889_pinout_v1_0.png?1457306365

Are they RST, CS and IRQ pins?

If so, even I tried all combinations between these 3 pins, why doesnt that work?

I am suffering here and requesting your little help @zrecore

zrecore, 2016-06-15:
Yes, by default. The Adafruit examples (such as the data mode example) use
hardware SPI with those pons as the default.

Without any idea of how your board is wired up, what firmware is on the BLE
module, what your code looks like, etc... I can't really help you further.

Pictures, code, video, etc. will help further investigate the issue.

zrecore, 2016-06-15:
... additionally, please mention what operating system you are using for
this, what Bluetooth hardware you have on your machine, etc. Details are
needed.

orcinz, 2016-06-15:
Ok. I am using Macbook Air & Windows 8 x64 with Bootcamp.
My bluetooth hardware is Apple Broadcom Built-in Bluetooth.

I connected an MPU-6050 sensor to my adafruit feather to receive X and Y variables.

I use that code;

```
#include <Arduino.h>
#include <SPI.h>
#if not defined (_VARIANT_ARDUINO_DUE_X_) && not defined(ARDUINO_ARCH_SAMD)
  #include <SoftwareSerial.h>
#endif

#include "Adafruit_BLE.h"
#include "Adafruit_BluefruitLE_SPI.h"
#include "Adafruit_BluefruitLE_UART.h"

#include "BluefruitConfig.h"

#include<Wire.h>
const int MPU_addr=0x68;  // I2C address of the MPU-6050
int16_t AcX,AcY;
int16_t BcX,BcY;
int16_t CcX,CcY;
int16_t i,x,y;

    #define FACTORYRESET_ENABLE         0
    #define MINIMUM_FIRMWARE_VERSION    "0.6.6"
/*=========================================================================*/

// Create the bluefruit object, either software serial...uncomment these lines
/*
SoftwareSerial bluefruitSS = SoftwareSerial(BLUEFRUIT_SWUART_TXD_PIN, BLUEFRUIT_SWUART_RXD_PIN);

Adafruit_BluefruitLE_UART ble(bluefruitSS, BLUEFRUIT_UART_MODE_PIN,
                      BLUEFRUIT_UART_CTS_PIN, BLUEFRUIT_UART_RTS_PIN);
*/

/* ...or hardware serial, which does not need the RTS/CTS pins. Uncomment this line */
// Adafruit_BluefruitLE_UART ble(BLUEFRUIT_HWSERIAL_NAME, BLUEFRUIT_UART_MODE_PIN);

/* ...hardware SPI, using SCK/MOSI/MISO hardware SPI pins and then user selected CS/IRQ/RST */
Adafruit_BluefruitLE_SPI ble(BLUEFRUIT_SPI_CS, BLUEFRUIT_SPI_IRQ, BLUEFRUIT_SPI_RST);

/* ...software SPI, using SCK/MOSI/MISO user-defined SPI pins and then user selected CS/IRQ/RST */
//Adafruit_BluefruitLE_SPI ble(BLUEFRUIT_SPI_SCK, BLUEFRUIT_SPI_MISO,
//                             BLUEFRUIT_SPI_MOSI, BLUEFRUIT_SPI_CS,
//                             BLUEFRUIT_SPI_IRQ, BLUEFRUIT_SPI_RST);

// A small helper
void error(const __FlashStringHelper*err) {
  Serial.println(err);
  while (1);
}

/**************************************************************************/
/*!
    @brief  Sets up the HW an the BLE module (this function is called
            automatically on startup)
*/
/**************************************************************************/
void setup(void)
{
  while (!Serial);  // required for Flora & Micro
  delay(500);

  Serial.begin(115200);
  Serial.println(F("Adafruit Bluefruit HID Keyboard Example"));
  Serial.println(F("---------------------------------------"));
[... 145 lines cut ...]
With that code, I connect my Adafruit Feather As Bluetooth HID Keyboard.

But in Adafruit Feather Library there is no option to configure my device as Bluetooth HID Gamepad.

I tried to use code shared here.

I downloaded the zip, copied library and examples to my arduino folder, opened HID Keyboard example and uploaded to my feather device.

Code uploaded without any errors.

As I open the serial, (in 115200 or 9600 tried both) there is nothing happening at the serial screen.

Then I search bluetooth devices with my PC and I see that my Adafruit Feather still shows itself as "Adafruit Bluetooth Device" 

I mean, uploading the code which I downloaded from here, doesnt change my device's bluetooth settings.

I thought that may because of Bluetooth pins, I tried many combinations, but couldnt succeeded..

I dont know what else I need to share with my experience, but these are all I can tell you about.

I just want to configure my adafruit feather as Bluetooth HID Gamepad. 

But couldnt succeeded for many days...

I hope you can help me with these informations I shared @zrecore

sandeepmistry, 2016-06-20:
@zrecore thanks for helping out!

@orcinz unfortunately this library only supports the Bluefruit LE based on the nRF8001. I think this is pretty clear in the [read me](https://github.com/sandeepmistry/arduino-BLEPeripheral#compatible-hardware). There was some discussion in #34 to support Adafruit board based on the nRF51822, however the conclusion wasn't possible to fully support it without running custom firmware.

However, if you're feeling adventurous you can try to load custom firmware on it using [arduino-nRF5](https://github.com/sandeepmistry/arduino-nRF5) with and external programmer and the nRF5 branch of this repo.

closed 2016-06-20 by sandeepmistry

Givy, 2017-12-02:
Hi,

I am trying to get the Adafruit Feather 32u4 Bluefruit LE working with RPi and I2C. How do I put it into Slave mode or I2C mode or what is the Slave address on this module?! I have it connected without the pull up resistors to the RPi. I have other components attached to RPi with I2C and when I run "i2cdetect -y 1" from RPi, I do see the slave addresses of the other components (see below), but none for the "Adafruit Feather 32u4 Bluefruit LE ". 

I also loaded the "Atcommand" sketch into Arduino and tried to use the "AT+HWI2CSCAN" from the serial line, but nothing comes back. 

/Giviprog# i2cdetect -y 1
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- 34 -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Thank you for your help.

Regards,

Givi