sandeepmistry/arduino-BLEPeripheral #70: Library not compatible with Redbearlab 1.0.6 Board library and new arduino ide

szotyi007, 2016-06-17:
I have upgraded arduino ide to 1.6.9, the redbearlab board to 1.0.6, the bleperipheral to 0.2.0 since then I m no longer able to upload.

Arduino/libraries/BLEPeripheral/nRF51822.cpp:8:19: fatal error: ble.h: No such file or directory
   #include <ble.h>

I have also tried to downgrade the Redbearlab board. In this case I do not get this:
Found programmer: Id = "RBL-nRF"; type = S
Programmer supports buffered memory access with buffersize=256 bytes.
Nothing else. Only the Done uploading.

sandeepmistry, 2016-06-20:
@szotyi007 things seem fine on my Mac using IDE 1.6.9 and a version of the RBL core from git:

```
commit b1bfff7e72b1b60f166804a82f2ac35e3aefb4ff
Author: Cheong <cheong@redbear.cc>
Date:   Wed Jan 27 12:18:48 2016 +0800
```

What OS are you using? Could you please share full build logs enabling verbose mode in the IDE?

punamdahiya, 2016-07-15:
I got the similar error as reported in this issue. I am on mac Yosemite version 10.10.4. Attaching build logs
[compile-error.txt](https://github.com/sandeepmistry/arduino-BLEPeripheral/files/365432/compile-error.txt)

punamdahiya, 2016-07-15:
commit b1bfff7e72b1b60f166804a82f2ac35e3aefb4ff RBL core is version 1.0.5 and uploads the code with the message 

Found programmer: Id = "RBL-nRF"; type = S
Programmer supports buffered memory access with buffersize=256 bytes.
Nothing else. Only the Done uploading. 

Updating the board manager to latest version 1.0.6 fails to compile with the error 
Arduino/libraries/BLEPeripheral/nRF51822.cpp:8:19: fatal error: ble.h: No such file or directory
#include

punamdahiya, 2016-07-15:
Some more findings, In 1.0.5
/Library/Arduino15/packages/RedBearLab/hardware/nRF51822/1.0.5/cores/RBL_nRF51822/BLE_API-master/ble/BLE.h

In 1.0.6 its changed to
/Library/Arduino15/packages/RedBearLab/hardware/nRF51822/1.0.6/cores/RBL_nRF51822/ble-master/ble

tigoe, 2016-07-22:
I'm getting the same error. It had been several months since I used the BLE Nano or the BLEPeripheral lib, so I removed all my old installs and started afresh.

* Using Arduino 1.6.9 on OSX
* Installed RedBear's 51822 board files using http://redbearlab.com/getting-started-nrf51822
* Installed BLEPeripheral using Library manager

Selected both BLE Nano 1.0 and 1.5. In both cases i get the same error when compiling any example that uses BLEPeripheral:  

````
/Users/tigoe/Documents/Arduino/libraries/BLEPeripheral/nRF51822.cpp:8:19: fatal error: ble.h: No such file or directory
   #include <ble.h>
                   ^
compilation terminated.
exit status 1
Error compiling for board BLE Nano(V1.0 16KB).
````
I'm guessing ble.h has been renamed, but haven't dug deeper yet.

tigoe, 2016-07-22:
Downgrading to 1.0.5 worked for me. See http://discuss.redbear.cc/t/troubleshooting-arduino-tutorial/569, apparently there's a 1.0.7 coming. Though I wonder if they've done some renaming of their files that might require an upgrade for this lib too.

sandeepmistry, 2016-07-25:
So for whatever reason RedBearLab renamed `arduino-1.6.x/hardware/RBL/RBL_nRF51822/cores/RBL_nRF51822/ble-nrf51822-master/source/nordic_sdk/components/softdevice/s130/headers/ble.h` to `arduino-1.6.x/hardware/RBL/RBL_nRF51822/cores/RBL_nRF51822/ble-nrf51822-master/source/nordic_sdk/components/softdevice/s130/headers/nrf_ble.h` in https://github.com/RedBearLab/nRF51822-Arduino/commit/b18d721da79ed784f54dd8a64a5b693535555568 which was released in the 1.0.6 release. Nordic's SDK has consistently named the header `ble.h`.

Let me ask them why this was done, and what their suggestion is moving forward. I'll open a new issue on their repo. Hopefully, it can get resolved in the 1.0.7 release.

In the short term, my suggestion is downgrading to v1.0.5 as @tigoe suggested.

sandeepmistry, 2016-07-25:
This will be fixed with the v1.0.7 release of the nRF51822-Arduino release: https://github.com/RedBearLab/nRF51822-Arduino/issues/70#issuecomment-234820401.

The change has already been made in master: https://github.com/RedBearLab/nRF51822-Arduino/commit/60d6139e11ab065b8c5af7d4bfbd75f90bc36fea.

sandeepmistry, 2016-08-14:
RedBear has released v1.0.7 of their nRF51 Arduino core, please upgrade to resolve this.

closed 2016-08-14 by sandeepmistry

tigoe, 2016-08-16:
thanks for the update!