sandeepmistry/arduino-BLEPeripheral #71: Callbacks for characteristic read operations

bluetooth-mdw, 2016-07-07:
I’d like the API to provide the ability to have callbacks relating to characteristic read operations. A common trick is to use the authorisation flag on characteristics so that all reads get intercepted and you can then do whatever you want in the implementation, returning true (i.e. authorised) at the end.

Read callbacks might be used to
- increment or recalculate a characteristic value "on demand"
- trigger a real time read from a sensor
- and so on....

Thanks

sandeepmistry, 2016-08-14:
Pull requests welcome for this, `nRF51822::poll()` is the place to hook in and trickle the event up.

Note, this feature will most likely be nRF51 specific and not work on the nRF80001.

bluetooth-mdw, 2016-08-15:
OK, thanks. I'll add it to my list :-)

baijupaul, 2016-10-20:
Would really love to see this feature to trigger a real time read of temperature. Thank you for all your efforts.