sandeepmistry/arduino-BLEPeripheral #72: Write a RemoteCharacteristic

andresrubiop, 2016-07-25:
Hello I am using the ANCS example and I am trying to write  the BLERemoteCharacteristic ancsControlPointCharacteristic for receive extra info in the ancsDataSourceCharacteristic.

Using this message.
struct AncsGetNotification
{
  unsigned char commandId;
  unsigned long notificationUid;
  unsigned char attributeId1;
  
};

What i am doing is to write just as the characteristic were a loca characteristic. So. Maybe can you guide me how to write remote characteristics.

Thanks.

sandeepmistry, 2016-08-14:
Hi @andresrubiop,

https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/remote_test/remote_test.ino#L90-L96 might be useful.

You should also be careful about the endianness of the `unsigned long notificationUid;` member of the structure. The Apple ANCS docs will tell you if it's big or little endian.

sandeepmistry, 2016-08-28:
I think this has been answered. I'm going to close it due to lack of activity.

closed 2016-08-28 by sandeepmistry