sandeepmistry/arduino-BLEPeripheral #73: GATT Client API

bluetooth-mdw, 2016-07-27:
It would be great if there was a class which provides a GATT client API. Some profiles have the GAP Peripheral as the GATT Client (e.g. the HTTP Proxy Service) so that once a connection has been accepted by the peripheral it then acts as GATT client and sends ATT PDUs over the connection to the GATT server at the other end which was previously the GAP Central device.

sandeepmistry, 2016-07-28:
I think want you want is already supported, checkout:
 * https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/remote_test/remote_test.ino
 * https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/remote_service/remote_service.ino (works with bleno's test.js)
 * https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/examples/ancs/ancs.ino

The API doc does not include this functionality at this time.

You'll have to know the UUID's you want to use upfront, because that's all the nRF8001 supports. Due to nRF8001 constraints this library will never support dynamic querying of remote characteristics. However, you never no if another library will be made to leverage all the features of the nRF51/52.

bluetooth-mdw, 2016-07-29:
Actually I was asking wrt Arduino 101 and unfortunately it looks like they don't have your "remote attribute" classes: https://github.com/01org/corelibs-arduino101/tree/master/libraries/CurieBLE/src 

I'll raise as an issue in that repo.

sandeepmistry, 2016-08-14:
Closing this than, this repo is specific to the nRF51 and nRF8001.

closed 2016-08-14 by sandeepmistry