sandeepmistry/arduino-BLEPeripheral #75: Blend Micro no visibility when creating a BLEPeripheral

withgallantry, 2016-07-29:
Hi,
Brilliant library, used it to make many a HID device. I've recently tried to get the test app from the library working via the Arduino IDE. 
I can get HID devices to be recognised by my iPhone (iOS9.3.3) no problem however when trying to use just BLEPeripheral.h to declare a BLE device it isn't recognised by any devices on iOS or Mac.
I'm using 1.0.3 RBL AVR Boards.
The code uploads correctly and the Serial outputs that it's initialised however I still can't see them.
Any suggestions appreciated.

withgallantry, 2016-07-29:
Very sorry, this issue is down to myself. I've since discovered that iOS doesn't show LE connections. Apologies.

closed 2016-07-29 by withgallantry

sandeepmistry, 2016-08-14:
@withgallantry I would suggest using the LightBlue app from the app store for testing.

More info. in the "Make: Bluetooth" book if you are interested. (http://makebluetooth.com).