sandeepmistry/arduino-BLEPeripheral #76: Write Long and Read Long

bluetooth-mdw, 2016-07-29:
Are these GATT sub-procedures supported? I don't seem to be able to specify these operations as properties when I create a BLECharacteristic i.e.

```
enum BLEProperty {
  BLEBroadcast         = 0x01,
  BLERead                 = 0x02,
  BLEWriteWithoutResponse = 0x04,
  BLEWrite                = 0x08,
  BLENotify               = 0x10,
  BLEIndicate            = 0x20
};
```

does not explicitly include values that represent this.

Needed for standard characteristics like URI: https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.uri.xml when used in the HTTP Proxy Service profile.

sandeepmistry, 2016-08-14:
Long reads are supported with using `BLEConstantCharacteristic` or `BLEProgmemConstantCharacteristic` - only constant characteristic values.

Do you have any examples of how to use long write on the nRF51? This feature is not likely to be supported on the nRF8001 though.

bluetooth-mdw, 2016-08-15:
Not that I know of. Maybe the later Nordic SDKs include an example. It seems to be documented in terms of sequence diagrams though: https://devzone.nordicsemi.com/documentation/nrf51/5.2.0/html/a00914.html

sandeepmistry, 2016-08-21:
@bluetooth-mdw thanks, the link you shared is for GATT client mode. Here's the one for GATT server mode: https://devzone.nordicsemi.com/documentation/nrf51/5.2.0/html/a00878.html

Pull requests well for this :)

shlomozippel, 2017-02-07:
The long writes require some extra state and memory management but long reads should just be handled seamlessly with the current logic, with the caveat that notify / indicate will only push out the first 20 bytes even if the characteristic is larger.

Currently I see that characteristic length is limited to 20 bytes. The actual allowed max length (on nrf51) is 510 or 512 bytes for fixed and variable length attributes.

This means that exposing larger values to be read from the peripheral should be relatively easy but letting values > 20 bytes to be written to is a larger change.