sandeepmistry/arduino-BLEPeripheral #77: Disconnect reason esp. supervisor timeout

bluetooth-mdw, 2016-08-11:
It would be great if it were possible to accompany disconnect events with a reason code. In particular, being able to distinguish between disconnections requested by the connected Central device and disconnections due to supervisor timeout being exceeded (e.g. device has gone out of range) would be useful if this is possible given the capabilities of different underlying platforms and APIs. If not all platforms this API supports provide this information then maybe a reason code of "unknown" could be used in those cases?

sandeepmistry, 2016-08-14:
@bluetooth-mdw do you have a purposed API for this feature?

bluetooth-mdw, 2016-08-15:
Not really. I suppose you could just add further values to BLEPeripheralEvent though.... if you keep BLEDisconnected = 1 meaning 'Disconnected, reason unknown" then code would remain compatible. Then you could add BLEDisconnectedSupervisorTimeout = 4 for example.