sandeepmistry/arduino-BLEPeripheral #80: Transmite Eddystone URI and UID data simultaneously

danielcbit, 2016-08-27:
Hi, I'm playing with Eddystone examples for nRF 8001 and I saw that in the current code you can transmit only one type of payload. How would I accomplish transmitting them "simultaneously"(change the advertising payload at each publish)?

sandeepmistry, 2016-08-28:
Hi @danielcbit,

I think you can try modifying the Eddystone beacon code to call toggle between the following
 * UID: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/EddystoneBeacon.cpp#L48-L64
 * URL: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/EddystoneBeacon.cpp#L73

after calling `begin`.

Since the advertising interval is 100ms, you'll have to toggle at a less frequent interval than this.

I'm closing this for now, feel free to continue the discussion though. Pull request welcome for the new example along with `setUid` API to make things cleaner.

closed 2016-08-28 by sandeepmistry

danielcbit, 2016-08-28:
Hi @sandeepmistry. I would have to toggle between those in the loop method, right?

sandeepmistry, 2016-08-28:
I would recommend it, two options for delaying are using:

 * `delay(ms)`
 * checking the value of `mills()` and comparing differences.

danielcbit, 2016-08-28:
Only mils() actually, since delay() stops the code execution so it is not a good idea.

Nice, I'll work on these patchs. I will add support for ESP8266 as well, I have a patch already. This is my current platform.

sandeepmistry, 2016-09-05:
@danielcbit sounds good, looking forward to seeing your patches.