sandeepmistry/arduino-BLEPeripheral #82: BleRead not read data from node.js

rusyabob, 2016-09-05:
```
// Import libraries (BLEPeripheral depends on SPI)
    #include <SPI.h>
    #include <BLEPeripheral.h>
    #include "BLESerial.h"
    
    // define pins (varies per shield/board)
    #define BLE_REQ   6
    #define BLE_RDY   7
    #define BLE_RST   4
    
    // create ble serial instance, see pinouts above
    BLESerial BLESerial(BLE_REQ, BLE_RDY, BLE_RST);
    
    
    void setup() {
      // custom services and characteristics can be added as well
      BLESerial.setLocalName("BlendMicro");
    
      Serial.begin(115200);
      BLESerial.begin();
    }
    
    void loop() {
      BLESerial.poll();
    
      forward();
      loopback();
      // spam();
    }
    
    
    // forward received from Serial to BLESerial and vice versa
    void forward() {
      if (BLESerial && Serial) {
        int byte;
        while ((byte = BLESerial.read()) > 0) Serial.write((char)byte);
        while ((byte = Serial.read()) > 0) BLESerial.write((char)byte);
      }
    }
    
    // echo all received data back
    void loopback() {
      if (BLESerial) {
        int byte;
        while ((byte = BLESerial.read()) > 0) BLESerial.write(byte);
      }
    }
    
    // periodically sent time stamps
    void spam() {
      if (BLESerial) {
        BLESerial.print(millis());
        BLESerial.println(" tick-tacks!");
        delay(1000);
      }
    }
```

I use this code on the BlendMicro. To send and read data i use this library for node.js https://www.npmjs.com/package/blendmicro . Connection RPI3<----->node.js<-------------->BlendMicro. BlendMIcro not read data from node.js. Node.js read data from BlendMicro. Please help me

sandeepmistry, 2016-09-05:
@vojtamolda any ideas on this one?

vojtamolda, 2016-09-06:
Hi guys,
I'll look into it. Please, give me a couple of days though. Things are quite busy these days...

sandeepmistry, 2016-09-06:
Thanks @vojtamolda!

vojtamolda, 2016-09-11:
Hello @rusyabob,

I'm having trouble reproducing your problem. It works for me when using the latest `BLEPeripheral` (`02dd9e` not `0.2.0`). Sketch code was exactly copy-pasted from your comment. I did tests on RFDuino (nRF51) and Arduino 101 (Curie) and both boards do the expected thing - sometimes send the BLE data back and sometimes forward BLE data to the serial port. It's also worth noting that I used an iMac/XCode Bluetooth Explorer and iPhone/Bluefruit App not a RPI3/node.js like you, but theoretically it shouldn't make any difference.

I'm not sure if "sometimes loopback and sometimes forward" is what you really want to do. More precisely "sometimes" depends on how lucky one is with timing the I/O versus the function executing on the board at that precise moment.

I think the root of the problem is that you're calling `loopback()` and `forward()` in sequence. The `Serial` and `BLESerial` are global objects and therefore are shared between the function calls. Each object encapsulates it's own buffer and depending on the timing of the I/O operation "sometimes" the data are received during `loopback()` and sometimes during `forward()`. It's hard to tell which scenario will happen.

To make your code do both `forward()` and `loopback()` at the same time you need to merge the functions together. For example like this:

```
void setup() {
  BLESerial.setLocalName("BlendMicro");
  Serial.begin(115200);
  BLESerial.begin();
}

void loop() {
  BLESerial.poll();
  forward_and_loopback();
}

void forward_and_loopback() {
  if (BLESerial && Serial) {
    int byte;
    while ((byte = BLESerial.read()) > 0) {
      BLESerial.write(byte);
      Serial.write((char)byte);
    }
    while ((byte = Serial.read()) > 0) {
      BLESerial.write((char)byte);
    }
  }
}
```

I guess the behavior and timing may be a bit different for nRF8001 boards (like BlendMicro), because of the underlying nRF8001 <--SPI--> processor connection. I don't have a board with nRF8001 available. Could you, please try the example above and get back to me?

rusyabob, 2016-09-11:
Hello! Thanks!
I conducted a test within a few days

воскресенье, 11 сентября 2016 г. пользователь Vojta Molda написал:

bilboquet, 2016-09-19:
Hi,
I commenting here because I'm facing an issue and I think it's the same as this one.
My configuration is as follow : 
- arduino M0 pro + RedBearLab BLE Shield V2.1
- PC running Linux
- Samsung GS4 with nRF UART v2.0

The PC is connected to the M0 through an usb cable on the programming port. Data are read from the M0 using `"cat /dev/ttyACM0"` and sent using `"echo 'xxx' > /dev/ttyACM0"`. Communication with the phone uses the nRF UART application.

What I have seen so far is that, using the unmodified serial.ino (so it should act as a bridge between the PC and the phone) communication works well from the PC to the phone (tested only with few characters) but, it fails from the phone to the PC. Let me give some details. If using the nRF UART app I try to send some data, I get an echo in the app but nothing on the PC. The sent data are displayed on the PC if I ever reprogram the board (some flush appears). If I modify the sketch, by adding a `Serial.println("");` after each `Serial.write((char)byte);` in the `forward()` function, the data find their way to the terminal of the PC but the echo is still present on the phone. Replacing `Serial.println("")`; by `Serial.flush();` does not work (i.e. the`flush()`seems to do nothing).
I'm still trying to solve this issue.
Regards,

PS: Please let me know if the comment is missplaced, in this case I'll open another issue.

vojtamolda, 2016-09-20:
Hello @bilboquet,

Thanks for reporting your problem. I tested a couple of sender _(RFDuino, Arduino 101)_ <--> receiver _(iMac/XCode Bluetooth Explorer, iPhone/Bluefruit, nRF Toolbox, LightBlue & BLE Tool Apps, RPI3/hcitool)_ combinations yesterday, but all the nRF51 boards I have run the unmodified `serial.ino` just fine.

Since there are now two people with the nRF8001 chips who have this problem I think it's worth investigating a bit more deeply. Please, give me a couple of days to get a hold of an nRF8001 somehow and I'll get back here with my findings.

sandeepmistry, 2016-09-21:
Thanks @vojtamolda, I can try to schedule some time to try this out on nRF8001 this weekend as well.

@rusyabob @bilboquet please provide a specific step by step set of instructions for either @vojtamolda  or me to try out.

bilboquet, 2016-09-22:
Hello,
I've investigated a little bit on my side.
My last finding might show that the problem is not related to BLESerial but to Serial.
On my M0 pro the following code keeps writing forever when it should not.
`void setup() {
  Serial.begin(115200);
  Serial.println("Starting !");
}

void loop() {
    int byte, count;
    while (count = Serial.available()) {
      delay(1000);
      Serial.print("data recv on Serial : ");
      Serial.println(count);

      byte = Serial.read();
      Serial.print((char)byte);
    }
}`
It behave like if there was a loop between read and write on Serial…

vojtamolda, 2016-09-25:
Hello @bilboquet,

Thanks for getting back here with your findings. I did some tests with _[Arduino Uno](https://www.arduino.cc/en/Main/ArduinoBoardUno) + [Adafruit nRF8001 Breakout](https://www.adafruit.com/products/1697)_ and I came to the same conclusion - everything works as it should.

There's an small issue that may be related to problems @rusyabob had, though. Being ignorant and unaware of the discussion in #28, I've decided to use the _Uno_ just for the heck of it and got a lesson in frugality in return 😃

I guess as a reflection of himself some ignorant developer wrote the`BLESerial` like huge memory hog 🐷 It contains a useless 256B RX buffer, which is empty all the time. Unless something crazy is going on, almost any processor can crunch the data much quicker than BLE can transfer. 256B doesn't make much of a difference on boards with 10s of kB of SRAM, but on smaller boards with 2kB or 2.5kB of SRAM it is a matter of literally life and death as the sketch barely fits.

I used the latest `0.3.0` version and wired the _Uno_ and _nRF8001_ together like this:
[image]

Pull-request to fix this is coming in a few minutes...

sandeepmistry, 2016-09-25:
Hi @vojtamolda,

Thanks for taking the time to look into this!

The Uno is a good test case because it only has 2k of RAM, the BLEPeripheral library is quite a RAM hog as well. The `BlendMicro` mentioned in https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/82#issue-175066618 has the same chip as the Leonardo, so has 2.5k of RAM.

I'm closing this for now, as I think @vojtamolda's pull request #85 fixes it.

@rusyabob please try it out and let us know of the results.

closed 2016-09-25 by sandeepmistry