sandeepmistry/arduino-BLEPeripheral #84: Question: Method for support features that is not available in all supported hardware

daviddedwin, 2016-09-15:
I am looking at #33 as the key method to support newer features that is present in some hardware but not in all supported hardware. Is this understanding correct ? 
Would it be possible to query the BLEDevice to know the current hardware and its supported feature set ?

sandeepmistry, 2016-09-21:
Hi @daviddedwin,

#33 is a bit unrelated, the root topic there is about no public API's to read the battery level and temperature.

Do you have specific new features in mind?

> Would it be possible to query the BLEDevice to know the current hardware and its supported feature set ?

It would be possible, however right now supported features on nRF8001 and nRF51822 are consistent.

sandeepmistry, 2016-10-15:
I'm closing this due to lack of activity.

closed 2016-10-15 by sandeepmistry

daviddedwin, 2016-10-17:
Will re-post issue when I have have proposal.