sandeepmistry/arduino-BLEPeripheral #88: Error compiled Arduino 1.6.11

sabas1080, 2016-10-05:
Hi

I have the next error in Arduino version 1.6.11

```
cc5DPaER.ltrans1.ltrans.o: In function `BLEUtil::addressToString(unsigned char const*, char*) [clone .constprop.39]':
cc5DPaER.ltrans1.o:(.text+0x218): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'
collect2: error: ld returned 1 exit status

```

In Arduino 1.6.9 all ok

Thanks

sandeepmistry, 2016-10-05:
I think this is a duplicate of #87 and has been fixed by https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/8f3aa2484cd97e48e14beb225413420b78e0ce42.

Can you please try out the latest version from master and let me know if it works? I can prep a new library manager this weekend.

closed 2016-10-05 by sandeepmistry

sabas1080, 2016-10-05:
Sorry, thanks