sandeepmistry/arduino-BLEPeripheral #90: Using interrupt with nRF8001.

bilboquet, 2016-10-07:
Hello, 

Looking into the 'utility' directory I've fallen upon the `#define HAL_ACI_TL_INTERRUPT` that enable an interrupt when RDYN line goes low.
I wonder if BLEPeripheral has been tested with this `#define` and if using it could increase system responsiveness and throughput on BLE?

Any feedback will be much appreciated.

Regards,

sandeepmistry, 2016-10-10:
@bilboquet could you clarify what you mean by " increase system responsiveness and throughput on BLE"?

That define would be equivalent to calling `attachInterrupt(...)` on the RDYN pin with a function that calls `blePeripheral.poll()` to handle events. Which would have a very similar behaviour to calling `blePeripheral.poll()` continuously in the `loop` of your sketch.

bilboquet, 2016-10-12:
Hello,
Thank you for the reply which confirms my understanding (and lets me think that using interrupt should work).
To explain what I mean by "increase system responsiveness and throughput", I have to say few words on what I'm trying to do. I'm actualy reading some values from a sensor and want to stream them as fast as possible through BLE. Actually it's not as quick as I would like and I'm even loosing some data. 
Hope my goal is clearer.
Regards,
Thank you for the reply which confirm what I was thinking

sandeepmistry, 2016-10-15:
You'll probably get better results by lowering the connection interval. You can also check the return values of `characteristic.setValue(...)` to see if it failed (this is where you are losing data).

https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#set-connection-interval

closed 2016-10-15 by sandeepmistry