sandeepmistry/arduino-BLEPeripheral #91: Appears advertising packet doesn't include ServiceList

scottjenson, 2016-10-10:
Recently pulled up an old RFDuino demo and it no longer works w/ Android! When I put the same URL into an Eddystone beacon, everything worked. Here is a hex dump of both devices, each with the same compressed URL: tiny.cc/cp9000

RFDuino
0x0201061416AAFE10EE0274696E792E63632F636239303030

Simple Eddystone beacon
0x0201060303AAFE1416AAFE10DE0274696E792E63632F636239303030

Notice that the RFDuino has two differences:
1) It's missing 0303AAFE in front which is the ServiceIDList parameter w/ the Eddystone ID 
2) Transmit powers are different (not relevant)

I'm assuming this is a simple change to the library to add this service list? Without this change, I'm afraid arduinos will no longer be found by the more picky Anrdoid Nearby scanner.

sandeepmistry, 2016-10-10:
Hi @scottjenson,

Thanks for pointing this out, there was a missing call to set the advertised service UUID in Eddystone URL mode. I've added it in https://github.com/sandeepmistry/arduino-BLEPeripheral/commit/8c4b05415108635bd111d148d546a4407c30d7a8.

I'll create a new tag for the library shortly so the fix will be available in the Arduino IDE's library manager.

closed 2016-10-10 by sandeepmistry

scottjenson, 2016-10-10:
Thanks for that. About how long will this take? I currently don't see it in the Manage Library... dialog but that's clearly a bit TOO early ;-) The reason I'm asking is that I'm giving a demo in 2 days. I was hoping to test tomorrow. If you think it'll easily be out by tomorrow, great.

sandeepmistry, 2016-10-10:
Somethings probably up with the Arduino library manager system, I don't see the new tag v0.3.1 tag in http://downloads.arduino.cc/libraries/library_index.json. I'll get someone to look at the system tomorrow.

In the meantime you can download https://github.com/sandeepmistry/arduino-BLEPeripheral/archive/0.3.1.zip and extract + replace the contents of `~/Documents/Arduino/libraries/BLEPeripheral`.

scottjenson, 2016-10-12:
> libraries/library_index.json. I'll get someone to look at the system
> BLEPeripheral/archive/0.3.1.zip and extract + replace the contents of

got it!

sandeepmistry, 2016-10-12:
Great, I forgot to mention that the Arduino library manager poller was fixed yesterday.