sandeepmistry/arduino-BLEPeripheral #97: Can this library work with RFdigital's Simblee nrf51822?

danlsk, 2016-10-29:
Please advice if this library can be compiled for Simblee. What are the changes necessary for test? I have the Simblee module here.

sandeepmistry, 2016-10-30:
Hi @danlsk,

Probably because it's based on nRF51, but it will require some work. To start find the references to `__RFduino__` in this library, and see what the equivalent define for Simblee is, then add a `|| defined(<simblee define>)`.

Then you'll have to somehow figure out what soft device version RFdigitial has flashed on it, and get the appropriate Nordic headers. For RFduino it was flashed with S110 v5 which is quite old, the headers for those are in [src/utility/RFduino](https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/master/src/utility/RFduino).

danlsk, 2016-10-31:
Hi  @sandeepmistry ,
  I have made some modifications to || defined(__Simblee__) . The board.txt seems to use __Simblee__.  I believe Simblee is using S130 soft device since it is able to perform both central and peripheral roles. There is no published soft device header for Simblee and I intend to ask RFdigital for this. But, meanwhile, how should I point to any current S130 headers? How should I copy Redbearlab one?

sandeepmistry, 2016-10-31:
> I believe Simblee is using S130 soft device since it is able to perform both central and peripheral roles.

That sounds right.

> There is no published soft device header for Simblee and I intend to ask RFdigital for this. 

Ok, let's see what they say. I've heard a lot of people being upset about them not open sourcing the RFduino code that was mentioned in the Kickstarter campaign.

> In fact, I was surprised how you sorted out one for RFduino. 

A bit of luck, just tried out a few versions of different S110 headers 😃 

> But, meanwhile, how should I point to any current S130 headers? How should I copy Redbearlab one?

RedBearLab might be too old. If you click on the "Downloads" tab on this page: https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF51822

There is an option to download S130 v1 and v2. Try both of those starting with v2. There are a few `#ifdef`'s in the `nRF51822.cpp/h` for RFduino that tie it to S110, you'll have to switch those to use S130 instead.

Also, we'll need to figure out if it has an onboard crystal or not, because the RFduino didn't, but all RedBearLab's board do.

Btw, is the Simble Arduino core available on Github somewhere?

danlsk, 2016-11-01:
The Simblee Arduino core is not available on Github. However, if you have installed the board support package by entering `https://www.simblee.com/package_simblee166_index.json` into the board support URL and installing it via board manager , you might be able to find the library headers in this installed folder below. 
`C:\Users\<your user folder>\AppData\Local\Arduino15\packages\Simblee\hardware\Simblee\1.1.0`

Is the library affected by the onboard crystal? RFduino and Simblee are System In Package that typically integrated the 32KHz crystals on package rather than  onboard. I am surprised you brought this up that the Raytac doesn't include this. Simblee is compiling and running at F_CPU 16MHz and not in overclocked mode. Is this the info you need on CPU clock speed?

I will get those nrf SDK library and dump into a library folder for compilation then. Will update you again.

danlsk, 2016-11-01:
1. I copied the S130 v2 header file into local \utility\Simblee
2. Copied nrf.h and compiler_abstraction.h into the folder too.
3. In nrf51822.h
`#elif defined(__Simblee__)
  #include <utility/Simblee/ble_gatts.h>
  #include <utility/Simblee/ble_gattc.h>
  #include <utility/Simblee/nrf_soc.h>`

I have this error during compilation.
`error "Device family must be defined. See nrf.h."`

sandeepmistry, 2016-11-01:
> Is the library affected by the onboard crystal? 

Just need to use the right option when calling `sd_softdevice_enable` in `nRF51822.cpp`.

> I have this error during compilation.

Try adding a `#define nRF51` before the includes Simblee include statements. I think the RFduino core has a `-DNRF51` C compiler flag in the core.

danlsk, 2016-11-01:
There is a build.bat at `C:\Users\<user folder>\AppData\Local\Arduino15\packages\Simblee\hardware\Simblee\1.1.0\system\Simblee\source` that already has this `-DNRF51`. Is this batch file ran during compilation? If so, it should be there somehow. But, it doesn't seem to get triggered.

sandeepmistry, 2016-11-06:
> Is this batch file ran during compilation?

Probably not. It probably builds a static library.

Did you try my suggestion for adding a `#define NRF51`? You can also change the `nrf.h` include in the headers to `nrf51.h` like I did for RFduino.

danlsk, 2016-11-06:
Adding the #define NRF51 in sketch. ```
C:\Users\......\Documents\Arduino\libraries\BLEPeripheral\src/utility/Simblee/nrf.h:51:38: fatal error: nrf51_deprecated.h: No such file or directory
#include "nrf51_deprecated.h"
```
Need to look for those library files.

sandeepmistry, 2016-11-06:
Add what about "You can also change the nrf.h includes in the headers to nrf51.h like I did for RFduino."

danlsk, 2016-11-06:
In which file did you do that for RFduino? I don't see that in nrf51822.h?

sandeepmistry, 2016-11-06:
See: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/src/utility/RFduino/nrf_sdm.h#L49-L53

It's including `nrf_sdm.h` not `nrf_soc.h` like you have in https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/97#issuecomment-257501703.

danlsk, 2016-11-06:
Made those changes in both `nrf_sdm.h` and `nrf_soc.h` like you did for RFduino in `/utility/RFduino`.
`#ifdef __Simblee__
#include "nrf51.h"
#else
#include "nrf.h"
#endif`

New error.

`C:\Users\danie_000\Documents\Arduino\libraries\BLEPeripheral\src\nRF51822.cpp: In member function 'virtual bool nRF51822::updateCharacteristicValue(BLECharacteristic&)':

C:\Users\danie_000\Documents\Arduino\libraries\BLEPeripheral\src\nRF51822.cpp:1047:116: error: cannot convert 'uint16_t* {aka short unsigned int*}' to 'ble_gatts_value_t*' for argument '3' to 'uint32_t sd_ble_gatts_value_set(uint16_t, uint16_t, ble_gatts_value_t*)'

sd_ble_gatts_value_set(localCharacteristicInfo->handles.value_handle, 0, &valueLength, characteristic.value());
`

sandeepmistry, 2016-11-13:
Take a look at the `NRF51_S130` define.

Can you push your work to a Github fork as well?

tolson2000, 2016-11-13:
Do you know of a low level call that would definitely determine if it is S130 or S110 that is in the Simblee?

danlsk, 2016-11-20:
@sandeepmistry I got the answer from RFdigital that Simblee is using S110 stack. I almost uploaded my files but this seems to be totally wrong since I was working on S130.

danlsk, 2016-11-29:
@sandeepmistry I got it compiled successfully using the RFduino S110 but the BLE stack call doesn't seem to work yet.

danlsk, 2016-12-01:
The RFduino S110 can still allow the Simblee module to boot with serial output. The V8 S110 has problem with that. I see that you work quite a lot on the S110 for RFduino because you put up with all those preprocessor. I believe you know better on integrating the S110 for Simblee.

sandeepmistry, 2016-12-02:
>  I believe you know better on integrating the S110 for Simblee.

Well I don't have a Simblee board ... closing for now then. Feel free to update your progress on the issue though.

closed 2016-12-02 by sandeepmistry

jjthrash, 2016-12-02:
I have a Simblee board I'm hoping to turn into a BT HID device, so I'm very interested in this issue. Don't really have any experience to contribute, though.