sandeepmistry/arduino-BLEPeripheral #98: How to send a string over characteristics

iqbalpalemad, 2016-10-31:
hai

i want to send a 18 character string  to my blend micro over  a characteristics. for this which characteristic i have to use . i have tried BLEFixedLengthCharacteristic but which shows an error unknown type name.So please tell me which characteristic is best suitable for transmitting a string...?

sandeepmistry, 2016-10-31:
@iqbalpalemad please see the second version of `setValue`: https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/API.md#set-value

I would recommend the generic `BLECharacteristic` with the max length of 20 for strings, unless the string will be fixed length.

In the future, please provide a minimal sample sketch that has the issue/question and compile output.

closed 2016-10-31 by sandeepmistry

mrdc, 2019-12-15:
@sandeepmistry Will you be so kind to show how to send a string properly?
When I use `BLEFixedLengthCharacteristic` I get this error:
`no matching function for call to 'BLEFixedLengthCharacteristic::BLEFixedLengthCharacteristic(const char [5], int)'`

TamojitSaha, 2020-02-25:
@sandeepmistry  can you help us ?