sandeepmistry/arduino-BLEPeripheral #102: Not able to resume after power cycle on an OTA bootloader

danlsk, 2016-11-09:
Using Redbearlab BLE NANO with OTA bootloader enabled, I am not able to resume the device name after the power cycle. The device became DFU_S130_V1.0 forever and never got out from OTA boot request. This issue however is not the case when I use the redbearlab library BLE_API.h. The OTA seems to timeout well and get back to the program. But using BLEPeripheral, I am stucked.

sandeepmistry, 2016-11-14:
@danlsk that function erases a flash sector (usually the last page).

danlsk, 2016-11-20:
I believe the flash sector might be a little different due to the OTA bootloader. It must have erased sketch/program vector address. The module is stucked at OTA blinking loop. The current bondstore.clearData() does not work correctly for the BLE Nano! If you have a BLE Nano there, you can definitely reproduce the error.

danlsk, 2016-11-23:
I am not sure how the clearData() or other things have deleted the OTA blink. I have to download the OTA bootloader again. So, now the clearData() and OTA work after power/reset cycle, using this change below.
`BLEBondStore bleBondStore(1);`

sandeepmistry, 2016-11-26:
Ok, so this means the last page of flash is used by the boot loader some how, passing 1 makes it use the second last page.

Feel free to open a pull request for the related examples to add a comment for RedBearLab nRF51 based boards should pass in one etc. Thanks.

danlsk, 2016-11-30:
I believe the BLEBondStore needs to define correctly. What i got may just be a quick workaround that may not subject to rigorous testing. I don't believe using your last page or my second last page is correct. nrf51 and nrf52 flash memory map is different also. [nrf5x flash memory map](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.0.0%2Flib_bootloader.html&cp=4_0_0_3_4_0_1&anchor=lib_bootloader_memory)
Also, RFduino doesn't have a reset button, hence, I didn't test the reset cycle. I may need a wire to test the reset pin. The power cycle on RFduino seems ok.

sandeepmistry, 2016-11-30:
I'm open to suggestions, please open a PR once you figure things out.

Not all boards are running the Nordic Bootloader - so the flash map does not necessarily apply. The RBL boards do use it though, RFduino from what I remember has a custom closed source boot loader.

danlsk, 2016-12-01:
What is the version of S110 that RFduino is using? I just want to double check the flash memory map. Anyhow, Your example works fine for RFduino without any changes. I tested the reset cycle. It behaves just like the power cycle. As for BLE Nano, I believe this code might be safer. 
NRF51
`BLEBondStore bleBondStore(21);`
NRF52
`BLEBondStore bleBondStore(32);`

sandeepmistry, 2016-12-02:
> What is the version of S110 that RFduino is using?

v5 with a custom (closed sourced) bootloader.

closed 2016-12-02 by sandeepmistry